project-chip/connectedhomeip #12650: Update (or small refactor) to AttributeValueEncoder

erjiaqing, 2021-12-07:
#### Problem

The AttributeValueEncoder is used for encode the attribute value field in attribute data ib. However, to support features like list chunking, it has to support encode a whole report ib instead of a single field.

Some changes need to be discussed:

- [ ] Field (names) in `AttributeValueEncoder::AttributeEncodeState`
- [ ] `mPath.mListOp` in EncodeList

This should probably taken place after #12053 

 #### Proposed Solution
<suggested fix, suggested enhancement>

erjiaqing, 2022-02-09:
The new title is not accurate (and even made me confused -- list chunking should be supported by AttributeValueEncoder), this should be a refactor and is not related to spec.

bzbarsky-apple, 2022-05-05:
@erjiaqing I think this issue could benefit from a clearer description of what the problem is, but I agree this is not a spec issue and not a 1.0 issue.

closed 2022-11-12 by stale[bot]