project-chip/connectedhomeip #12733: Develop work around in endpoint_config.zapt for "attribute accessor" storage class

jonsmirl, 2021-12-08:
Zap does not currently support defining storage classes for attributes. While this is on the road map to get fixed in Zap, a temporary solution is needed. 

The temporary solution is to add dedicated code in endpoint_config.zapt which will detect the Matter attributes implemented via the "attribute accessor" storage class and generate the correct code.   (A way to flag this would be to look for type='ARRAY" and no length specified). 

Later when Zap catches up this code can be removed.

bzbarsky-apple, 2021-12-08:
Note to self: Need to do this for things with array and struct types.

bzbarsky-apple, 2022-04-21:
Update:  This is done for lists in #13846.

For structs, code generation will fail at the moment for non-external structs, though the error will be a little inscrutable.  A better fix is blocked on https://github.com/project-chip/zap/issues/391

Furthermore, we have a way of flagging attributes as "must have external storage" in the SDK, in `src/app/zap-templates/zcl/zcl.json`.

So not sure how much is left to do here immediately.

closed 2022-06-13 by bzbarsky-apple