project-chip/connectedhomeip #12762: Do we have two dns caches?

cecille, 2021-12-09:
#### Problem
Looks like the case manager introduced one, and we also have one in the platform impl.

 #### Proposed Solution
We should have one or even none - in the case of platform we might actually have three.

turon, 2022-01-12:
It would be helpful to add link references that jump to where the two implementations are in the code.  Also, could extra clarification be added about why and how some platforms may have three mdns caches in the proposed solution?  I believe the three implementations is expanding on the problem as platform packages such as avahi add a third cache implementation, but that is unclear in the bug description.

cecille, 2022-01-13:
Third cache would be the internal cache in the platform impl (not always there - depends on the impl). If you were running avahi, the dns result would be cached in the avahi impl, the platform cache and the controller cache. 

SDK cache impl is src/lib/dnssd/DnssdCache.h. It's instantiated in both CHIPDeviceController.cpp and Discovery_ImplPlatform.

see https://github.com/project-chip/connectedhomeip/blob/master/src/controller/CHIPDeviceController.h line 341 and
https://github.com/project-chip/connectedhomeip/blob/master/src/lib/dnssd/Discovery_ImplPlatform.cpp line 42.

If we're going to vote one off the island, the one in Discovery_ImplPlatform has the least utility IMO. Though perhaps I'm missing some context from the original author. See #11632

holbrookt, 2022-01-31:
@cecille seems like this could be needed for 1.0?

cecille, 2022-02-01:
Could be - it's wasteful at the very least.

woody-apple, 2022-02-08:
Let's defer until we know this is required?

andy31415, 2022-02-24:
Assigned to me as I believe I am touching this as part of mDNS resolution logic (I need to have a 'get best address' logic and the DNS caching is actually in the way in parts). Not working to really fix it, but it may be touched as a sideffect.

andy31415, 2022-05-11:
Removed caching.

closed 2022-05-11 by andy31415