project-chip/connectedhomeip #12884: Media playback cluster preconditions and commands failling

manjunathambiger23, 2021-12-10:
Summary:
              As per the test-plan media playback cluster precondition and commands seems to fail when we execute

Test-plan screenshot:
[image]

Spec media playback commands screenshot:
[image]

Yaml script screenshot:
[image]

Execution logs:
[image]
[Precondition_fail.txt](https://github.com/project-chip/connectedhomeip/files/7692983/Precondition_fail.txt)
[command_fail.txt](https://github.com/project-chip/connectedhomeip/files/7693158/command_fail.txt)

turon, 2022-02-01:
Playback cluster must pass cert tests for v1.
Removing tag `v1_triage_split_4`

manjunath-grl, 2022-05-04:
This issue was created by me using a account that i had to delete due to access issues, I will continue to follow these issues with my current id-  manjunath-grl

closed 2022-05-25 by woody-apple