project-chip/connectedhomeip #12927: ZLL Clusters implementation broken ?

caipiblack, 2021-12-11:
#### Problem

I need to use the `OnWithTimedOff` command from the OnOff cluster,

This functionality seems to be available only in zll-on-off-server.c, if i include this file to the build of my project I get errors during the build because of an include that is not pointing to the correct folder.

```
#include « ../../include/af.h »
```

As `af.h` file is in the `src/app/util` folder, it can’t find the include.

Is it supposed to build ?

 #### Proposed Solution

fix includes and build errors for all the zll cluster implementations

caipiblack, 2021-12-11:
I just realize that there is another issue about it: https://github.com/project-chip/connectedhomeip/issues/12018#issue-1058182987

franck-apple, 2022-01-21:
Closing this one, please refer to #12018

closed 2022-01-21 by franck-apple