project-chip/connectedhomeip #12929: Enable Github Discussions

ayushsharma82, 2021-12-12:
#### Problem
I wanted to know more about how to set up a matter controller. My problem is that there is no forum!  There is hardly any doc available outside this repository. It's too bad for a beginner to hop into an active community of "code only" rather than forms where I can search for my issue and understand the topic in "human language" rather than looking through readme(s) in each directory.

 #### Proposed Solution
Please enable GitHub "discussions" on this repository so that people can ask & share information with each other. It will serve justice to people interested in getting started with it.

woody-apple, 2022-01-21:
Thank you for the suggestion, we will considering this moving forward.

closed 2022-10-18 by stale[bot]

olalonde, 2023-03-07:
@woody-apple Could you enable discussions? Or maybe launch a Discord server? I too have questions related to Matter that are not really code issues.