project-chip/connectedhomeip #12944: OTA-Requestor-App fails on building Idf.py build command using master branch with ESP-32 with this commit id -f85d6ed38757e13fe3dad10e34b69c65fb46da67

sowmyassp, 2021-12-13:
Problem
  While using the OTA-Requestor-app on ESP-32 with the following command,its failed with building
 * expected behavior - build done
 * actual behavior - fail
 * Commands used - idf.py build
 * Actual logs attached 
[OTA-Requestor-app fails on esp-32 in idf.py build.odt](https://github.com/project-chip/connectedhomeip/files/7705180/OTA-Requestor-app.fails.in.idf.py.build.odt)

shubhamdp, 2021-12-14:
There has been a lot of refactoring in the common code which has broken the ota-requestor-app/esp32. #12804 also adds a much needed refactoring and it would be good to refactor example once that gets merge. For time being you can use all-clusters-app/esp32 to try out OTA.

cjandhyala, 2021-12-14:
@shubhamdp do you have a timeline when this refactoring is expected to be fixed ?

shubhamdp, 2021-12-15:
@cjandhyala #13017 PR fixes this.

dhrishi, 2022-01-18:
Closing as the linked PR is merged. @sowmyassp Please feel free to re-open in case this does not fix your problem

closed 2022-01-18 by dhrishi