project-chip/connectedhomeip #12945: Bridge app fails on ESP-32 using master branch with this commit id - f85d6ed38757e13fe3dad10e34b69c65fb46da67

sowmyassp, 2021-12-13:
problem
 While using the bridge app on esp-32 with the following commands,its failed with this attribute cluster-revision
 * expected behavior - success
 * actual behavior  - failure 
 * commands used- ./chip-tool  bridgeddevicebasic read cluster-revision 1 0
 * actual logs attached -
 
[bridge app fail in esp-32.odt](https://github.com/project-chip/connectedhomeip/files/7705572/bridge.app.fail.in.esp-32.odt)

chiragatal, 2021-12-14:
The bridge-app adds the bridged_device_basic clusters on only the dynamic endpoints i.e. 2, 3, 4, 5, 6 (out of which 3 is then removed in the example). If you try to run the command with endpoints 2, 4, 5, 6, it should work.

dhrishi, 2021-12-14:
Thanks @chiragatal . @sowmyassp Please try it out and close the issue if it works fine.

sowmyassp, 2021-12-14:
Its working fine.

closed 2021-12-14 by sowmyassp