project-chip/connectedhomeip #12952: Automatically handle persistent storage of data backed by `AttributeAccessInterface`

mrjerryjohns, 2021-12-13:
#### Problem

This PR exposes a very real issue in the SDK where-by every `AttributeAccessInterface` implementor is hand-rolling their own persistent storage implementation to store to NVM pertinent attribute data

The above PR highlights a problem that provides a window into the proposed solution - TLV data coming off the wire is first de-serialized into a cluster-object, then adaptered to flow through the data provider interface, before finally, being converted _back_ to TLV at the time of persistent storage.

This is wasteful, and results in 2x the code needed to handle the TLV serialization/de-serialization.

### Proposed Solution

Utilizing cluster-objects, automatically 'extract' out the data from the interface using a `Read()` into TLV, and write that TLV data directly to NVM. This can happen at any time Read() is called (e.g when something has changed).

When a `Write()` is invoked, the TLV data will also then be written into NVM.

In this regard, the `AttributeAccessInterface` implementor is only responsible for the volatile view of the data, while an equivalent copy is kept in NVM.

On subsequent boot, the data is read out from NVM and blitted to each and every provider by calling their respective `Write` methods. This will then 're-seed' their starting state correctly.

Couple of key benefits:
1. Re-use of cluster objects and TLV to act as both an over-the-wire format as well as a persistent storage format.
2. Cluster-objects work really well for versioning - if a cluster gets a new attribute that isn't in storage, the defaulting treatment will automatically initialize a default value correctly.
3. No API changes needed to existing call-sites.

mrjerryjohns, 2021-12-13:
FYI @bzbarsky-apple @mlepage-google @tcarmelveilleux

mlepage-google, 2021-12-13:
Note that AccessControl cluster implementation does not use cluster-objects for serialize/deserialize. Instead, it serializes/deserializes directly into access control entries. Is that going to be a problem?

mlepage-google, 2021-12-13:
FYI @harimau-qirex