project-chip/connectedhomeip #12987: After device reboot, chip-device-ctrl can't communicate with the device

caipiblack, 2021-12-14:
#### Problem

[image]

Using the `chip-device-ctrl`, after successful pairing we can control a Matter object but if the object reboot we can't communicate with him anymore.

**Steps to reproduce:**

1. Use  `chip-device-ctrl` to pair the device and control the on/off cluster:
```
ble-scan
connect -ble 3842 20202021 1
close-ble
resolve 1
zcl OnOff Off 1 1 1
zcl OnOff On 1 1 1
```
2. Reboot the device
3. In the same `chip-device-ctrl` window, try again to control the on/off cluster of the device
```
zcl OnOff Off 1 1 1
zcl OnOff On 1 1 1
```
4. The packet is dropped in the device side and there are timeouts in  `chip-device-ctrl`

**chip-device-ctrl logs:**

```
chip-device-ctrl > zcl OnOff Off 1 1 1
[1639486338.282969][86381:86387] CHIP:IN: Prepared encrypted message 0x1714a18 to 0x0000000000000001 (1)  of type 0x8 and protocolId (0, 1) on exchange 7266i with MessageCounter:3.
[1639486338.283038][86381:86387] CHIP:IN: Sending encrypted msg 0x1714a18 with MessageCounter:3 to 0x0000000000000001 (1) at monotonic time: 18517277 msec
[1639486343.250253][86381:86387] CHIP:IN: Sending encrypted msg 0x1714a18 with MessageCounter:3 to 0x0000000000000001 (1) at monotonic time: 18522245 msec
[1639486343.252646][86381:86387] CHIP:DL: Avahi resolve found
[1639486343.252733][86381:86387] CHIP:DIS: Node ID resolved for 0x0000000000000001
[1639486343.252758][86381:86387] CHIP:DIS:     Addr 0: [192.168.1.41]:5540
[1639486343.252776][86381:86387] CHIP:CTL: OperationalDiscoveryComplete for device ID 0x0000000000000001
Commissioning complete
[1639486343.252937][86381:86387] CHIP:CTL: Address resolved for node: 0x0000000000000001
Node address has been updated
[1639486343.504758][86381:86387] CHIP:IN: Sending encrypted msg 0x1714a18 with MessageCounter:3 to 0x0000000000000001 (1) at monotonic time: 18522499 msec
[1639486343.761600][86381:86387] CHIP:IN: Sending encrypted msg 0x1714a18 with MessageCounter:3 to 0x0000000000000001 (1) at monotonic time: 18522756 msec
[1639486344.017509][86381:86387] CHIP:EM: Failed to Send CHIP MessageCounter:3 on exchange 7266i sendCount: 3 max retries: 3
[1639486350.285598][86381:86387] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 7266i
An exception occurred during process ZCL command:
Chip Stack Error 50
chip-device-ctrl >
```

**esp32 logs:**

```
E (12378) chip[IN]: Data received on an unknown connection (2). Dropping it!!
E (17348) chip[IN]: Data received on an unknown connection (2). Dropping it!!
E (17598) chip[IN]: Data received on an unknown connection (2). Dropping it!!
E (17858) chip[IN]: Data received on an unknown connection (2). Dropping it!!
```

retfie, 2021-12-19:
I see the same behavior with lock_app example on nrf52840dk over openthread. Probably credentials are not stored into permanent storage, but only into device RAM and they are gone after reboot. This is just a hypotheses, and I also would like to know the root cause.
Thanks !

dhrishi, 2021-12-19:
@caipiblack I think it is not yet supported in chip-device-ctrl. Instead you can use [chip-tool](https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool) for commissioning which facilitates this. Please give it a try.

caipiblack, 2021-12-20:
Thanks for your responses.

In fact I am building a commissioner based on `chip-tool`, but it works more like `chip-device-ctrl` (the stack event loop keep running during all the software execution) so I have the same problems as `chip-device-ctrl` in such cases.

I think the problem is not an issue with the storage in device side because if I reboot the software (controller), the session correctly opens without any pairing (similar to `chip-tool`).

I made some debug to see what is the difference between:

- Trying to send messages to a device that just reboot
- Rebooting the software (commissioner) after the device reboot and trying to send messages to the device

What I see is:

In the function `connect()` of `OperationalDeviceProxy.cpp`, if the software (commissionner) is rebooted the device is seen as `State::Initialized`, then it will re-establish the session with the device but if the commissionner is still alive (when the device reboot) it continue to see the session as `State::SecureConnected` so it try to send the message to the device, but the device reject it. There is a timeout in commissioner side.

So now I have some questions:

- If this is the normal process, how the commissionner is supposed to make the difference between a device that is not connected and a device that reject the messages because the session is not established (after reboot) ?
- What is the supposed behaviour of the commissioner in such case ?

gjc13, 2022-01-11:
This is typically caused by `chip-device-ctrl` holding a stale CASE connection.
Try using `close-seession` to terminate the CASE session and using `resolve` command to connect again.

caipiblack, 2022-01-11:
> This is typically caused by `chip-device-ctrl` holding a stale CASE connection. Try using `close-session` to terminate the CASE session and using `resolve` command to connect again.

Yep, that's what I think but, why the stack doesn't close the session ? How the stack is supposed to work in such case ?

I don't see anything about this in the specifications.

As I am building my own commissioner, I would like to know if I have to close the session myself or not. 

The error is detected on the failure callback when we execute a command on a cluster:

```
chip::Controller::OnOffCluster cluster;
cluster.Associate(device, command->mEndpointId);
cluster.On(command->mOnSuccessCallback.Cancel(), command->mOnFailureCallback.Cancel());
```

The afStatus returned is `EMBER_ZCL_STATUS_FAILURE`, this is a very generic error so I'm not sure it's very good to call `ReleaseOperationalDevice(command->mNodeId)` from `DeviceController` every time we get this error.

About your suggest: I don't check with `chip-device-ctrl`, but if I call `ReleaseOperationalDevice()` in my software when the error happens, it close the session, so the next request will re-open the session and everything is good.

vivien-apple, 2022-01-27:
This is unclear to me that it is test harness or v1.0 related ?

@caipiblack summary seems correct where the controller "think" the connection to the device is still alive and tries to send messages without reestablishing the secure session.
I'm not sure how this case should be handled from the controller side - that said I can reproduce that using YAML and I have to manually close the session for it to work.

@mrjerryjohns  @bzbarsky-apple Do you know what is the expected behaviour ?

bzbarsky-apple, 2022-01-28:
Ideally we would do either a notification to the app (which can then attempt to re-establish a connection) or an automaticattempt to re-establish on in parallel with MRP retries once we detect non-delivery of an MRP message.

If that's on the 1.0 feature list, it needs to get staffed.  There's work that would have to be done to make it happen.

vivien-apple, 2022-02-02:
It is unclear to me that this stuff fails under the SDK requirements from test harness.
Basically it works with `chip-tool` since `chip-tool` reestablish the connection each time.
It work with `YAML` if you issue a `Reboot` step and then a `waitForCommissionee` step, since it  close/re-open the connection.
So while I agree this is an issue I'm going to remove the `sdk requirements from test harness` as it is distracting for the work I am currently doing.

CC @woody-apple

woody-apple, 2022-02-08:
I agree, tagging as post V1.0

caipiblack, 2022-05-17:
Update: This problem is still present in TE9 commit. Now it can be easily re-produced with chip-tool as it include an "interactive mode"

caipiblack, 2022-06-07:
On TE9 the problem seems to be the same as before with some changes.

On the initial commit where I made this issue, I was able to bypass the problem by calling `CurrentCommissioner().ReleaseOperationalDevice(this->mNodeId);` from my code in case of request failure with a device.

Like that, on the next request the stack was re-opening the session with the device and voila.

But now, on TE9 commit (and maybe newer commits), even if we call `CurrentCommissioner().ReleaseOperationalDevice(this->mNodeId);`, when we arrive in the function `Connect()` of `OperationalDeviceProxy.cpp`:

- `mState` = `State::NeedsAddress`
- `AttachToExistingSecureSession()` returns `true` **so it's like the session is still there**.

So we still have the problem.

Also, note that the initial problem is easy to reproducible with chip-tool, as it supports "interactive mode":

```
./chip-tool interactive start

...wait for the stack to be ready...
...reset your device and wait for the device to be ready for commissioning...

>>> pairing qrcode-wifi 20 SSID PASSWORD MT:-24J042C00KA0648G00

...wait for the commissioning to finish...

>>> onoff toggle 20 1

...the command works...
...reboot the device...

>>> onoff toggle 20 1

...it fails as the packets are dropped by the device...

```

Errors on device side:

```
E (15098) chip[IN]: Data received on an unknown session (LSID=19021). Dropping it!
E (20838) chip[IN]: Data received on an unknown session (LSID=19021). Dropping it!
```

This problem is a real pain for controllers. 

I understand that you will look about this problem only after v1.0 and that it's possibly not clear how the stack is supposed to treat the problem. 

But for now, did someone know a new way to bypass the problem ? Why `ReleaseOperationalDevice()` is not enough ?

caipiblack, 2022-06-15:
For information when using TE9 commit (and probably next commits) we are able to bypass the problem by calling the function `DisconnectDevice(NodeId)` from `CHIPDeviceController.cpp` in case of timeout during a request.

Then as we arrive to close the session, the next request will re-open it.

But, this function is not future proof: 

```
    /**
     * TEMPORARY - DO NOT USE or if you use please request review on why/how to
     * officially support such an API.
     *
     * This was added to support the 'reuse session' logic in cirque integration
     * tests however since that is the only client, the correct update is to
     * use 'ConnectDevice' and wait for connect success/failure inside the CI
     * logic. The current code does not do that because python was not set up
     * to wait for timeouts on success/fail, hence this temporary method.
     *
     * TODO(andy31415): update cirque test and remove this method.
     *
     * Returns success if a session with the given peer does not exist yet.
     */
```

So please, do not remove the function until the problem is solved. @andy31415 

Thanks,

caipiblack, 2022-07-04:
Update: 

For those who bypass the problem by closing the session with a device in case of timeout. I find a case where is not really good:

[image]

plan44, 2022-07-13:
Controller code as present in AppleTV 16 beta 3 (I guess, something past TE9) still has the problem. I can successfully commission the bridge-app via an iPad, which then passes control to the AppleTV and everything works fine.

However, when the bridge-app is restarted, the AppleTV keeps trying to connect with a stale session getting the bridge app to log `CHIP: [IN] Data received on an unknown session (LSID=xxx). Dropping it!`, about once a minute, apparently "forever" (until I reboot the AppleTV). Only after reboot, the controller resumes communication.

From the discussion above I am not sure if there is a mechanism in place already that should detect, close and re-establish stale sessions already, or if this is a pending issue (for after V1.0?)
 
Sorry if this is the wrong place to ask this.

bzbarsky-apple, 2022-07-15:
@plan44 the mechanism is not in place yet, but should be for v1, we hope.

noname202, 2022-11-14:
Are there any updates on this? Can confirm that the behavior described by @plan44 is still present with the current beta software 16.2 for the HomePod Mini (Tested with the all-clusters-app example for ESP32). Unfortunately I didn‘t have the time to work through the specification, so I can’t comment on weather there are any issues with the all—clusters example app or the current controller implementation on the HomePod. However, in my opinion there needs to be a mechanism for detecting and re-establishing stale sessions in place. The current state of things makes for an unreliable user experience.

plan44, 2022-11-14:
> @noname202 Can confirm that the behavior described by @plan44 is still present with the current beta software 16.2 for the HomePod Mini

No, not any more.

In my testing, with a [heavily modified/extended bridge-app](https://github.com/plan44/p44mbrd), the problem disappeared in a later 16.1 beta and is definitely no longer present in 16.1 public release nor in current 16.2 beta.

Restarting the bridge *does* first receive a few attempts to connect the previous session (`CHIP: [IN] Data received on an unknown session (LSID=xxx). Dropping it!` log message) but after a short while the controller establishes a new session and stops using the stale one.

So if you still see the problem it must be something slightly different.

caipiblack, 2022-11-14:
With the SDK 1.0.0 this problem is solved with chip-tool in interactive mode.

If there is no objection i’m going to close this issue

noname202, 2022-11-15:
> No, not any more.
>
> In my testing, with a [heavily modified/extended bridge-app](https://github.com/plan44/p44mbrd), the problem disappeared in a later 16.1 beta and is definitely no longer present [...]

I just checked and you’re in right. After a few attempts the controller stops using the stale session. However, in many cases it doesn’t establish a new session. Instead the device (ESP32 all-clusters-app) does not log any connection attempts from the HomePod at all and the HomePod needs to be restarted. Chip-tool on the other hand works just fine. Therefore the behavior probably isn’t related to this issue.

HiFiPhile, 2022-12-08:
>With the SDK 1.0.0 this problem is solved with chip-tool in interactive mode. 
If there is no objection i’m going to close this issue

Please brought the fix to `chip-repl`

On `commit ab48bfcfadca66c7f1675f29b5d5ee5d35f10494 (HEAD -> v1.0-branch, origin/v1.0-branch)` ReadAttribute fails when MAD is rebooted:
```
In [2]: await devCtrl.ReadAttribute(2, [2])
2022-12-08 15:06:37 pi4-zli chip.DMG[14074] ERROR Time out! failed to receive report data from Exchange: 46502i
---------------------------------------------------------------------------
ChipStackError                            Traceback (most recent call last)
Cell In[2], line 1
----> 1 await devCtrl.ReadAttribute(2, [2])

File /home/pi/code/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py:1156, in ChipDeviceController.ReadAttribute(self, nodeid, attributes, dataVersionFilters, returnClusterObject, reportInterval, fabricFiltered, keepSubscriptions)
   1119 async def ReadAttribute(self, nodeid: int, attributes: typing.List[typing.Union[
   1120     None,  # Empty tuple, all wildcard
   1121     typing.Tuple[int],  # Endpoint
   (...)
   1129     typing.Tuple[int, typing.Type[ClusterObjects.ClusterAttributeDescriptor]]
   1130 ]], dataVersionFilters: typing.List[typing.Tuple[int, typing.Type[ClusterObjects.Cluster], int]] = None, returnClusterObject: bool = False, reportInterval: typing.Tuple[int, int] = None, fabricFiltered: bool = True, keepSubscriptions: bool = False):
   1131     '''
   1132     Read a list of attributes from a target node, this is a wrapper of DeviceController.Read()
   1133
   (...)
   1154         When not provided, a read request will be sent.
   1155     '''
-> 1156     res = await self.Read(nodeid, attributes=attributes, dataVersionFilters=dataVersionFilters, returnClusterObject=returnClusterObject, reportInterval=reportInterval, fabricFiltered=fabricFiltered, keepSubscriptions=keepSubscriptions)
   1157     if isinstance(res, ClusterAttribute.SubscriptionTransaction):
   1158         return res

File /home/pi/code/connectedhomeip/out/python_env/lib/python3.9/site-packages/chip/ChipDeviceCtrl.py:1117, in ChipDeviceController.Read(self, nodeid, attributes, dataVersionFilters, events, returnClusterObject, reportInterval, fabricFiltered, keepSubscriptions)
   1115 if res != 0:
   1116     raise self._ChipStack.ErrorToException(res)
-> 1117 return await future

ChipStackError: Chip Stack Error 50

In [3]:
```

What even worse is it leads to Segmentation fault even if devCtrl.CloseSession(2) is called trying to handle the exception.

In a production environment this unexpected crashing behavior is unacceptable since MAD could reboot at anytime. 

```
In [4]: devCtrl.CloseSession(2)
Out[4]: 0

In [5]:

In [5]: await devCtrl.ReadAttribute(2, [2])
2022-12-08 15:07:26 pi4-zli chip.DL[14074] ERROR Avahi resolve found
Segmentation fault
```

caipiblack, 2024-06-13:
> With the SDK 1.0.0 this problem is solved with chip-tool in interactive mode.
> 
> If there is no objection i’m going to close this issue

I close the issue for that reason.

closed 2024-06-13 by caipiblack