project-chip/connectedhomeip #12994: In raspi to raspi chip-tool pairing fails on OTA software updater command  on master branch with this commit id -c8027a5f50b10409b108f1d58d990f9b8f6bd435

sowmyassp, 2021-12-14:
Problem - while using the otasoftwareupdater command in chiptool on raspi to raspi its get failed .
 * expected behavior - success
 * actual behavior - fail
 * Actual command - sudo ./chip-tool otasoftwareupdaterequestor announce-ota-provider 1 0 0 4321 1
 * Actual logs attached here
   fails on  ota-chip tool command
[ota chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/7712989/ota.chip-tool.txt)
[ota-provider.txt](https://github.com/project-chip/connectedhomeip/files/7712994/ota-provider.txt)
[ota-requestor.txt](https://github.com/project-chip/connectedhomeip/files/7712999/ota-requestor.txt)

selissia, 2021-12-14:
@sowmyassp  : What does the network look like? Is it three separate Raspberry PIs on a single Ethernet segment? WiFi? 
Has this setup ever worked?

carol-apple, 2021-12-14:
From the logs, looks like usage of `announce-ota-provder` may be incorrect:

The chip-tool was used to commission the provider app with node id 1234:
```
$ sudo ./chip-tool pairing onnetwork 1234 20202021
```

The chip-tool was used to commission the requestor app with node id 4321
```
$ sudo ./chip-tool pairing onnetwork-long 4321 20202021 3840
```

The chip-tool was used to issue `announce-ota-provider` command to requestor with node id 4321, with provider node id 1 and endpoint 1
```
$ sudo ./chip-tool otasoftwareupdaterequestor announce-ota-provider 1 0 0 4321 1
```
The first value after `announce-ota-provider` should be the provider node id which is 1234, also the last value is the endpoint-id which should be 0.

carol-apple, 2021-12-19:
Test team validated that with expected command, this failure reported is not seen. Filed https://github.com/project-chip/connectedhomeip/issues/11943 to handle bad values in a better way.

closed 2021-12-19 by carol-apple