project-chip/connectedhomeip #13013: Only call MatterReportingAttributeChangeCallback on write if the data version changes.

bzbarsky-apple, 2021-12-14:
#### Problem
Right now we have two places where writes can call `MatterReportingAttributeChangeCallback`:

1. `WriteSingleClusterData`
2. `emAfWriteAttribute`

Both call it unconditionally, even if the value being written is identical to the current value.

 #### Proposed Solution
Once we have dataversion support, read the dataversion, then write the value, then read the dataversion and only report if it has changed.

bzbarsky-apple, 2023-10-16:
Note: the implementation proposed above does not work, because we actually update data version from inside MatterReportingAttributeChangeCallback nowadays.

tcarmelveilleux, 2023-10-16:
Overall: this is the largest cause of needless reports, and fixing this will have a big impact on reducing congestion for large fan-outs of lighting commands.