project-chip/connectedhomeip #13024: YAML: Support to read/write bit/bits (mask) to/from bitmap

jmeg-sfy, 2021-12-15:
#### Problem
Some tests relies on bit flag behaviors
Many cluster (still) use old fashion registers flags set/get and declare bitmap (even FEATURE MAP btw)

We need the ability to test those attribute by reading and/or writing a particular bit or bits via a mask or something in the YAML

This would be a blocker for tests and certain test step automation
 #### Proposed Solution
Idk but we need to figure this out

bzbarsky-apple, 2021-12-15:
Trying to understand the scope here.  Is this looking for something that basically does "read the current value, set or clear a given bit, write the result"?

jmeg-sfy, 2021-12-15:
yes , 
- 1 . the ability to read a bit for a simple test success or failure. 
- 2 And also to write a particular bit, without changing the others to trigger a flags for example

bzbarsky-apple, 2021-12-15:
One sort-of-possible approach:
1. Use the saveAs bits.
2. Add some commands that perform bit operations

So you could read, saveAs, then do your bit fiddling, saveAs, then write the saved value.

andy31415, 2022-05-31:
#19008 looks into at least type definition support

we can expand yaml tests (underlying bits or masked values) later if that seems useful