project-chip/connectedhomeip #13029: Need YAML support for accessing elements form saved lists

manjunathambiger23, 2021-12-15:
Summary:          
              We are trying to save an attribute that is of type list using saveAs keyword and we need to access once of the elements from that list.. something like "savedListVariable[0]", the PR #11798 demos only basic data types like integers and does not talk about complex data types like lists and objects, which are returned by some commands.

Test-plan Screenshot:
[image]

Execution screenshot:
[image]

bzbarsky-apple, 2022-01-07:
This is quite hard to do in general, because for non-numeric types the value is referencing memory in the TLV response that will get deallocated.  So the only way to save it is to clone the memory somehow, etc....

woody-apple, 2022-01-26:
This is going to be significant work, for we believe few test cases, suggest lowering priority on this one.

woody-apple, 2022-03-03:
Given the low priority, moving this out of V1.