project-chip/connectedhomeip #13044: Deprecate SessionManager::FindSecureSessionForNode()

sharadb-amazon, 2021-12-15:
#### Problem
FindSecureSessionForNode() is being used for legacy tests which use nodeId to send packets and for the tv-casting-app that uses the TV's node ID to find the associated secure session. It is based on using NodeId to locate the SecureSession.

 #### Proposed Solution
Deprecate this API and all its usages in the SDK/examples with an alternate API that would use PeerID.

bzbarsky-apple, 2022-05-05:
In #17266 this code was updated to no longer use back-door APIs, but now it's assuming sync callbacks from an operation that might be async (CASE establishment).

bzbarsky-apple, 2022-05-05:
That said, this is a problem for this specific example; it's not a spec issue or an app cluster issue.

bzbarsky-apple, 2022-05-05:
The sync behavior is still there; it's just moved to `TargetVideoPlayerInfo::Initialize`

chrisdecenzo, 2022-05-05:
ok, the Initialization returns an error in the async case but the callback correctly completes the init. This code could be cleaned up but this is not a launch blocker.

bzbarsky-apple, 2022-05-06:
> but this is not a launch blocker

Agreed; hence the label changes above.