project-chip/connectedhomeip #13075: LogEvent with 0x000000000000010D would fail with error Invalid TLVTag during eventReportBuilder.CreateEventData()

yunhanw-google, 2021-12-16:
#### Problem
LogEvent with 0x000000000000010D would fail with error Invalid TLVTag during eventReportBuilder.CreateEventData()

./gn_build.sh in macbook, then run ./out/debug/mac_x64_clang/chip-im-responder

wait for EventNumber 0x000000000000010C and 0x000000000000010D, then it would generate the error with Invalid TLVTag, later logEvent would be fine.

turon, 2022-02-01:
Should add unit test to verify this strange corner case has been fixed in v1.
Removing tag `v1_triage_split_4`

yunhanw-google, 2022-02-21:
when LogEvent has filled up the circular event  buffer,  https://github.com/project-chip/connectedhomeip/blob/master/src/app/EventManagement.cpp#L495, writer.Init(*mpEventBuffer); would fail where write would initialize circularEventBuffer, which further call CHIPCircularTLVBuffer::OnInit and call GetNewBuffer https://github.com/project-chip/connectedhomeip/blob/master/src/lib/core/CHIPCircularTLVBuffer.cpp#L261, and further fail in https://github.com/project-chip/connectedhomeip/blob/master/src/lib/core/CHIPCircularTLVBuffer.cpp#L202 because of mQueueLength >= mQueueSize, since we have set alwaysFail for mProcessEvictedElement in https://github.com/project-chip/connectedhomeip/blob/master/src/app/EventManagement.cpp#L157, EvictHead would always fail

yunhanw-google, 2022-02-21:
we probably skip the error instead of alwaysFail, or mark it as expected.

PR #15683 (2022-03-02): Fix issue 13075
#### Problem
* Fixes #13075, 
When mQueueLength >= mQueueSize happens in CHIPCircularTLVBuffer::GetNewBuffer, if mProcessEvictedElement is set with AlwaysFail, EvictHead would fail when logging Event

#### Change overview
--We should let EvictHead and Init succeed, then later EnsureSpaceInCircularBuffer
would help get enough space for new event in LogEventPrivate.
--Fix the urgent flag set in EventManagement encoding.
#### Testing
Manual validating with chip-im-responder via generating huge amount of events for hours. 
Add unit test to generate huge amount of events to reach th [... cut]
closed 2022-03-02 by andy31415