project-chip/connectedhomeip #13084: Command SkipChannel is failling

manjunathambiger23, 2021-12-16:
Summary:
                SkipChannel commands seems to fail with response as failure instead of success when we execute.
Spec screenshot:
[image]

Screenshot and log: 
[image]
[skipchannel.txt](https://github.com/project-chip/connectedhomeip/files/7728447/skipchannel.txt)

bzbarsky-apple, 2022-01-07:
@lazarkov @chrisdecenzo

lazarkov, 2022-02-07:
I've tested the master branch all is good now.

One terminal tab to run chip-tv-app:
`➜  connectedhomeip git:(master) ./out/debug/standalone/chip-tv-app`

One terminal tab to run chip-tool read command:
`➜  connectedhomeip git:(master) out/debug/standalone/chip-tool channel skip-channel-request  1 1 1`

Here's the response:

[1644278071841] [99515:1068681] CHIP: [EM] Received message of type 0x9 with protocolId (0, 1) and MessageCounter:449925 on exchange 47355i
[1644278071841] [99515:1068681] CHIP: [EM] Found matching exchange: 47355i, Delegate: 0x7ff731affec0
[1644278071841] [99515:1068681] CHIP: [EM] Rxd Ack; Removing MessageCounter:12886860 from Retrans Table on exchange 47355i
[1644278071841] [99515:1068681] CHIP: [EM] Removed CHIP MessageCounter:12886860 from RetransTable on exchange 47355i
[1644278071841] [99515:1068681] CHIP: [DMG] ICR moving to [ResponseRe]
[1644278071841] [99515:1068681] CHIP: [DMG] InvokeResponseMessage =
[1644278071841] [99515:1068681] CHIP: [DMG] {
[1644278071841] [99515:1068681] CHIP: [DMG] 	suppressResponse = false, 
[1644278071841] [99515:1068681] CHIP: [DMG] 	InvokeResponseIBs =
[1644278071841] [99515:1068681] CHIP: [DMG] 	[
[1644278071841] [99515:1068681] CHIP: [DMG] 		InvokeResponseIB =
[1644278071841] [99515:1068681] CHIP: [DMG] 		{
[1644278071841] [99515:1068681] CHIP: [DMG] 			CommandStatusIB =
[1644278071841] [99515:1068681] CHIP: [DMG] 			{
[1644278071841] [99515:1068681] CHIP: [DMG] 				CommandPathIB =
[1644278071841] [99515:1068681] CHIP: [DMG] 				{
[1644278071841] [99515:1068681] CHIP: [DMG] 					EndpointId = 0x1,
[1644278071841] [99515:1068681] CHIP: [DMG] 					ClusterId = 0x504,
[1644278071841] [99515:1068681] CHIP: [DMG] 					CommandId = 0x3,
[1644278071841] [99515:1068681] CHIP: [DMG] 				},
[1644278071841] [99515:1068681] CHIP: [DMG] 				
[1644278071841] [99515:1068681] CHIP: [DMG] 				StatusIB =
[1644278071841] [99515:1068681] CHIP: [DMG] 				{
[1644278071841] [99515:1068681] CHIP: [DMG] 					status = 0x0,
[1644278071841] [99515:1068681] CHIP: [DMG] 				},
[1644278071841] [99515:1068681] CHIP: [DMG] 				
[1644278071841] [99515:1068681] CHIP: [DMG] 			},
[1644278071841] [99515:1068681] CHIP: [DMG] 			
[1644278071841] [99515:1068681] CHIP: [DMG] 		},
[1644278071841] [99515:1068681] CHIP: [DMG] 		
[1644278071841] [99515:1068681] CHIP: [DMG] 	],
[1644278071841] [99515:1068681] CHIP: [DMG] 	
[1644278071841] [99515:1068681] CHIP: [DMG] 	InteractionModelRevision = 1
[1644278071841] [99515:1068681] CHIP: [DMG] },
[1644278071841] [99515:1068681] CHIP: [DMG] Received Command Response Status for Endpoint=1 Cluster=0x0000_0504 Command=0x0000_0003 Status=0x0
[1644278071841] [99515:1068681] CHIP: [DMG] ICR moving to [AwaitingDe]
[1644278071841] [99515:1068681] CHIP: [EM] Sending Standalone Ack for MessageCounter:449925 on exchange 47355i
[1644278071841] [99515:1068681] CHIP: [IN] Prepared secure message 0x70000be58560 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 47355i with MessageCounter:12886861.
[1644278071841] [99515:1068681] CHIP: [IN] Sending encrypted msg 0x70000be58560 with MessageCounter:12886861 to 0x0000000000000001 (1) at monotonic time: 49726244 msec
[1644278071842] [99515:1068681] CHIP: [EM] Flushed pending ack for MessageCounter:449925 on exchange 47355i
[1644278071842] [99515:1068674] CHIP: [CTL] Shutting down the commissioner
[1644278071842] [99515:1068674] CHIP: [CTL] Shutting down the controller
[1644278071842] [99515:1068674] CHIP: [CTL] Shutting down the commissioner
[1644278071842] [99515:1068674] CHIP: [CTL] Shutting down the controller
[1644278071842] [99515:1068674] CHIP: [CTL] Shutting down the commissioner
[1644278071842] [99515:1068674] CHIP: [CTL] Shutting down the controller
[1644278071842] [99515:1068674] CHIP: [CTL] Shutting down the commissioner
[1644278071842] [99515:1068674] CHIP: [CTL] Shutting down the controller
[1644278071842] [99515:1068674] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1644278071842] [99515:1068674] CHIP: [DL] Inet Layer shutdown
[1644278071842] [99515:1068674] CHIP: [DL] System Layer shutdown

closed 2022-02-07 by lazarkov

manjunathambiger23, 2022-02-08:
@vivien-apple we need YAML support for "SkipChannel" command which is showing failure response when we execute, the above which lazarkov told its not related to YAML

bzbarsky-apple, 2022-02-08:
@manjunathambiger23 It should work if you run the YAML against tv-app, no?  all-clusters-app does not have a working implementation of the TV clusters, and no one is planning to add one.