project-chip/connectedhomeip #13087: Not able to store List using saveAs keyword

manjunathambiger23, 2021-12-16:
Summary:
            Not able to store the List using saveAs keyword.
YAML screenshot:
[image]

Execution log and screenshot:
[image]
[SaveAs_list.txt](https://github.com/project-chip/connectedhomeip/files/7727839/SaveAs_list.txt)

woody-apple, 2022-01-26:
We believe this blocks a small number of test cases, likely will de-prioritize.

manjunathambiger23, 2022-02-01:
we are facing type error when we remove disable tag, previously it was failing while building 
[image]
[image]

woody-apple, 2022-03-03:
As far as well know this is only blocking a couple tests, suggest moving to post V1.