project-chip/connectedhomeip #13128: ASAN failure during unit tests storing fabric info

bzbarsky-apple, 2021-12-17:
https://github.com/project-chip/connectedhomeip/runs/4558860959?check_suite_focus=true shows an ASAN failure.  It's based on master commit 9338395fb503eb3df4bdc40c28ea987de0e5f582 in terms of the code inspection below.

 #### Problem
<details>
<summary>ASAN output:</summary>

```
2021-12-17 10:20:44.983 ERROR   10:20:41.489 - APP  ERR  : AddressSanitizer:DEADLYSIGNAL
2021-12-17 10:20:44.983 ERROR   10:20:41.489 - APP  ERR  : =================================================================
2021-12-17 10:20:44.983 ERROR   10:20:41.489 - APP  ERR  : ==40593==ERROR: AddressSanitizer: SEGV on unknown address 0x000253000048 (pc 0x7fff2031841f bp 0x7000054d6470 sp 0x7000054d61d8 T1)
2021-12-17 10:20:44.983 ERROR   10:20:41.489 - APP  ERR  : ==40593==The signal is caused by a READ memory access.
2021-12-17 10:20:44.983 ERROR   10:20:44.129 - APP  ERR  :     #0 0x7fff2031841f in objc_release+0x1f (libobjc.A.dylib:x86_64+0x841f)
2021-12-17 10:20:44.983 ERROR   10:20:44.795 - APP  ERR  :     #1 0x105798943 in chip::DeviceLayer::PersistedStorage::KeyValueStoreManager::Put(char const*, void const*, unsigned long) KeyValueStoreManager.h:222
2021-12-17 10:20:44.983 ERROR   10:20:44.795 - APP  ERR  :     #2 0x10579808c in chip::Server::ServerStorageDelegate::SyncSetKeyValue(char const*, void const*, unsigned short) Server.h:108
2021-12-17 10:20:44.983 ERROR   10:20:44.795 - APP  ERR  :     #3 0x105798582 in chip::Server::ServerStorageDelegate::SyncStore(unsigned char, char const*, void const*, unsigned short) Server.h:122
2021-12-17 10:20:44.984 ERROR   10:20:44.796 - APP  ERR  :     #4 0x105798778 in non-virtual thunk to chip::Server::ServerStorageDelegate::SyncStore(unsigned char, char const*, void const*, unsigned short) Server.h
2021-12-17 10:20:44.984 ERROR   10:20:44.796 - APP  ERR  :     #5 0x1056e44d9 in chip::FabricInfo::CommitToStorage(chip::FabricStorage*) FabricTable.cpp:108
2021-12-17 10:20:44.984 ERROR   10:20:44.796 - APP  ERR  :     #6 0x1056eaafd in chip::FabricTable::Store(unsigned char) FabricTable.cpp:450
2021-12-17 10:20:44.984 ERROR   10:20:44.797 - APP  ERR  :     #7 0x10543f1e2 in emberAfOperationalCredentialsClusterAddNOCCallback(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::app::Clusters::OperationalCredentials::Commands::AddNOC::DecodableType const&) operational-credentials-server.cpp:454
2021-12-17 10:20:44.984 ERROR   10:20:44.797 - APP  ERR  :     #8 0x105495d4f in chip::app::Clusters::OperationalCredentials::DispatchServerCommand(chip::app::CommandHandler*, chip::app::ConcreteCommandPath const&, chip::TLV::TLVReader&) IMClusterCommandHandler.cpp:1265
2021-12-17 10:20:44.984 ERROR   10:20:44.797 - APP  ERR  :     #9 0x10549bfdb in chip::app::DispatchSingleClusterCommand(chip::app::ConcreteCommandPath const&, chip::TLV::TLVReader&, chip::app::CommandHandler*) IMClusterCommandHandler.cpp:1899
2021-12-17 10:20:44.984 ERROR   10:20:44.797 - APP  ERR  :     #10 0x1054ccde5 in chip::app::InteractionModelEngine::DispatchCommand(chip::app::CommandHandler&, chip::app::ConcreteCommandPath const&, chip::TLV::TLVReader&) InteractionModelEngine.cpp:598
2021-12-17 10:20:44.984 ERROR   10:20:44.798 - APP  ERR  :     #11 0x1054af712 in chip::app::CommandHandler::ProcessCommandDataIB(chip::app::CommandDataIB::Parser&) CommandHandler.cpp:288
2021-12-17 10:20:44.984 ERROR   10:20:44.798 - APP  ERR  :     #12 0x1054ad9e3 in chip::app::CommandHandler::ProcessInvokeRequest(chip::System::PacketBufferHandle&&, bool) CommandHandler.cpp:130
2021-12-17 10:20:44.984 ERROR   10:20:44.798 - APP  ERR  :     #13 0x1054ac79f in chip::app::CommandHandler::OnInvokeCommandRequest(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&, bool) CommandHandler.cpp:79
2021-12-17 10:20:44.984 ERROR   10:20:44.799 - APP  ERR  :     #14 0x1054c8419 in chip::app::InteractionModelEngine::OnInvokeCommandRequest(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&, bool, chip::Protocols::InteractionModel::Status&) InteractionModelEngine.cpp:281
2021-12-17 10:20:44.984 ERROR   10:20:44.799 - APP  ERR  :     #15 0x1054cae91 in chip::app::InteractionModelEngine::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&) InteractionModelEngine.cpp:420
2021-12-17 10:20:44.984 ERROR   10:20:44.799 - APP  ERR  :     #16 0x1055b2c53 in chip::Messaging::ExchangeContext::HandleMessage(unsigned int, chip::PayloadHeader const&, chip::Transport::PeerAddress const&, chip::BitFlags<chip::Messaging::MessageFlagValues, unsigned int>, chip::System::PacketBufferHandle&&) ExchangeContext.cpp:486
2021-12-17 10:20:44.984 ERROR   10:20:44.799 - APP  ERR  :     #17 0x1055bb959 in chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SessionHandle const&, chip::Transport::PeerAddress const&, chip::SessionMessageDelegate::DuplicateMessage, chip::System::PacketBufferHandle&&) ExchangeMgr.cpp:308
2021-12-17 10:20:44.984 ERROR   10:20:44.799 - APP  ERR  :     #18 0x1055fd74a in chip::SessionManager::SecureUnicastMessageDispatch(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) SessionManager.cpp:535
2021-12-17 10:20:44.984 ERROR   10:20:44.800 - APP  ERR  :     #19 0x1055fc51c in chip::SessionManager::OnMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) SessionManager.cpp:378
2021-12-17 10:20:44.985 ERROR   10:20:44.800 - APP  ERR  :     #20 0x10560d287 in chip::TransportMgrBase::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) TransportMgrBase.cpp:71
2021-12-17 10:20:44.985 ERROR   10:20:44.800 - APP  ERR  :     #21 0x10571f9fa in chip::Transport::Base::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) Base.h:102
2021-12-17 10:20:44.985 ERROR   10:20:44.801 - APP  ERR  :     #22 0x10571f0db in chip::Transport::UDP::OnUdpReceive(chip::Inet::UDPEndPoint*, chip::System::PacketBufferHandle&&, chip::Inet::IPPacketInfo const*) UDP.cpp:120
2021-12-17 10:20:44.985 ERROR   10:20:44.801 - APP  ERR  :     #23 0x105569164 in chip::Inet::UDPEndPointImplSockets::HandlePendingIO(chip::BitFlags<chip::System::SocketEventFlags, unsigned char>) UDPEndPointImplSockets.cpp:688
2021-12-17 10:20:44.985 ERROR   10:20:44.801 - APP  ERR  :     #24 0x10556753b in invocation function for block in chip::Inet::UDPEndPointImplSockets::BindImpl(chip::Inet::IPAddressType, chip::Inet::IPAddress const&, unsigned short, chip::Inet::InterfaceId) UDPEndPointImplSockets.cpp:218
2021-12-17 10:20:44.985 ERROR   10:20:44.842 - APP  ERR  :     #25 0x105f31b5a in __wrap_dispatch_source_set_event_handler_block_invoke+0xca (libclang_rt.asan_osx_dynamic.dylib:x86_64+0x43b5a)
2021-12-17 10:20:44.985 ERROR   10:20:44.889 - APP  ERR  :     #26 0x7fff202ce805 in _dispatch_client_callout+0x7 (libdispatch.dylib:x86_64+0x3805)
2021-12-17 10:20:44.985 ERROR   10:20:44.889 - APP  ERR  :     #27 0x7fff202d11af in _dispatch_continuation_pop+0x1a6 (libdispatch.dylib:x86_64+0x61af)
2021-12-17 10:20:44.985 ERROR   10:20:44.889 - APP  ERR  :     #28 0x7fff202e1563 in _dispatch_source_invoke+0x80c (libdispatch.dylib:x86_64+0x16563)
2021-12-17 10:20:44.985 ERROR   10:20:44.889 - APP  ERR  :     #29 0x7fff202d4492 in _dispatch_lane_serial_drain+0x106 (libdispatch.dylib:x86_64+0x9492)
2021-12-17 10:20:44.985 ERROR   10:20:44.889 - APP  ERR  :     #30 0x7fff202d50ac in _dispatch_lane_invoke+0x16d (libdispatch.dylib:x86_64+0xa0ac)
2021-12-17 10:20:44.985 ERROR   10:20:44.889 - APP  ERR  :     #31 0x7fff202dec0c in _dispatch_workloop_worker_thread+0x32a (libdispatch.dylib:x86_64+0x13c0c)
2021-12-17 10:20:44.985 ERROR   10:20:44.904 - APP  ERR  :     #32 0x7fff2047545c in _pthread_wqthread+0x139 (libsystem_pthread.dylib:x86_64+0x345c)
2021-12-17 10:20:44.985 ERROR   10:20:44.904 - APP  ERR  :     #33 0x7fff2047442e in start_wqthread+0xe (libsystem_pthread.dylib:x86_64+0x242e)
2021-12-17 10:20:44.985 ERROR   10:20:44.904 - APP  ERR  : 
2021-12-17 10:20:44.985 ERROR   10:20:44.905 - APP  ERR  : ==40593==Register values:
2021-12-17 10:20:44.985 ERROR   10:20:44.905 - APP  ERR  : rax = 0x000000025300002e  rbx = 0x00007000054d6280  rcx = 0x0000000253000028  rdx = 0x00000001057bfc00  
[... cut ...]
2021-12-17 10:20:44.986 ERROR   10:20:44.905 - APP  ERR  : SUMMARY: AddressSanitizer: SEGV (libobjc.A.dylib:x86_64+0x841f) in objc_release+0x1f
2021-12-17 10:20:44.986 ERROR   10:20:44.906 - APP  ERR  : Thread T1 created by T0 here:
2021-12-17 10:20:44.986 ERROR   10:20:44.906 - APP  ERR  :     <empty stack>
2021-12-17 10:20:44.986 ERROR   10:20:44.906 - APP  ERR  : 
2021-12-17 10:20:44.986 ERROR   10:20:44.906 - APP  ERR  : ==40593==ABORTING
2021-12-17 10:20:44.986 ERROR   ================ CAPTURED LOG END ====================
```

</details>

 #### Proposed Solution
Not sure yet.  Trying to figure out which value was garbage....

bzbarsky-apple, 2022-06-28:
The relevant code has changed significantly, so not sure how useful this is at this point.

@tcarmelveilleux thoughts?

woody-apple, 2022-11-02:
Issue Scrub: Per @tcarmelveilleux we don't believe this is an issue anymore.

closed 2022-11-02 by woody-apple