project-chip/connectedhomeip #13129: Lighting-app stopped building for nRF52840-dongle

caubut-charter, 2021-12-17:
Lighting app stopped building on the nRF52840-dongle 2 days ago.  Timing correlates with the nRF Connect SDK 1.8 SDK merge (https://github.com/project-chip/connectedhomeip/pull/13031).

Build environment is a github action worker run nightly.  Issue occurs at this step in the guide following using the docker instructions: https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nrfconnect#building
```
west build -b nrf52840dongle_nrf52840
```

It appears some kconfig keys may have changed from the error.

**Error message:**
```console
...
Parsing /var/ncs/zephyr/Kconfig
/var/chip/examples/lighting-app/nrfconnect/boards/nrf52840dongle_nrf52840.conf:22: warning: attempt to assign the value 'y' to the undefined symbol USB
Loaded configuration '/var/ncs/zephyr/boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840_defconfig'

/var/chip/examples/lighting-app/nrfconnect/boards/nrf52840dongle_nrf52840.conf:23: warning: attempt to assign the value 'y' to the undefined symbol USB_UART_CONSOLE

/var/chip/examples/lighting-app/nrfconnect/boards/nrf52840dongle_nrf52840.conf:26: warning: attempt to assign the value '"CDC_ACM_0"' to the undefined symbol UART_SHELL_ON_DEV_NAME

error: Aborting due to Kconfig warnings

Merged configuration '/var/chip/config/nrfconnect/app/sample-defaults.conf'
Merged configuration '/var/chip/examples/lighting-app/nrfconnect/prj.conf'
Merged configuration '/var/chip/examples/lighting-app/nrfconnect/boards/nrf52840dongle_nrf52840.conf'
CMake Error at /var/ncs/zephyr/cmake/kconfig.cmake:270 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /var/ncs/zephyr/cmake/app/boilerplate.cmake:546 (include)
  /var/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /var/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:52 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/var/chip/examples/lighting-app/nrfconnect/build -S/var/chip/examples/lighting-app/nrfconnect -GNinja -DBOARD=nrf52840dongle_nrf52840
```

caubut-charter, 2021-12-17:
Supporting documentation for the depreciation of these options:

https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/releases/release-notes-2.7.rst
```
* Revised configuraiton of USB device support. Removed Kconfig option CONFIG_USB and introduced Kconfig option CONFIG_USB_DEVICE_DRIVER to enable USB device controller drivers, which is selected when option CONFIG_USB_DEVICE_STACK is enabled.
* Removed CONFIG_UART_SHELL_ON_DEV_NAME Kconfig option in favor of direct use of chosen node zephyr,shell-uart.
```

https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/releases/release-notes-3.0.rst
```
* Removed Kconfig option CONFIG_USB_UART_CONSOLE. Option CONFIG_USB_UART_CONSOLE was only relevant for console driver when CDC ACM UART is used as backend. Since the behavior of the CDC ACM UART is changed so that it more closely mimics the real UART controller, option is no longer necessary.
```

caubut-charter, 2021-12-17:
Possible solution: https://github.com/project-chip/connectedhomeip/compare/master...caubut-charter:update-lighting-nrf52840-dongle-sdk1.8

It looks to build now; need to test on an actual device.

lmpprk, 2021-12-29:
Hi, I have been also trying to resolve the same issue, I thought I fixed it but didn't see your investigation about the CI until now.

I tried the CI script and also constantly got the error ``"undefined reference to `__device_dts_ord_74'"``. But I then found out that if the CI script is run with the overlay config then it would build successfully. 

`scripts/examples/nrfconnect_example.sh lighting-app nrf52840dongle_nrf52840 -DOVERLAY_CONFIG=boards/nrf52840dongle_nrf52840.conf`

caubut-charter, 2021-12-31:
> You can try this: [master...lmpprk:nrf-dongle](https://github.com/project-chip/connectedhomeip/compare/master...lmpprk:nrf-dongle)

Thanks @lmpprk!  I will test after the holidays.  Good catch on updating the IFDEF's and CI script.  Should hopefully fix the issue I'm having over at https://github.com/caubut-charter/matter-rpi4-nRF52840-dongle/actions for generating nightly builds.

PR #13285 (2022-01-10): Fix lighting-app build error for nrf52840dongle
#### Problem
What is being fixed? 
* Fix #13129, detailed below:

The nrf light example app gets error when building for `nrf52840dongle_nrf52840` with following unrecognized symbols in kconfig

> `CONFIG_USB` -> [replaced](https://docs.zephyrproject.org/latest/releases/release-notes-2.7.html#usb) by `CONFIG_USB_DEVICE_DRIVER ` in 2.7
>`CONFIG_UART_SHELL_ON_DEV_NAME` -> [removed](https://docs.zephyrproject.org/latest/releases/release-notes-2.7.html?highlight=config_uart_shell_on_dev_name#changes-in-this-release) in 2.7 in favor of chosen node `ze [... cut]
closed 2022-01-10 by andy31415