project-chip/connectedhomeip #13146: Improper exchange context lifetime handling in PASESession.cpp and CASESession.cpp

msandstedt, 2021-12-19:
#### Problem

`PASESession` and `CASESession` set `mExchangeContext` to nullptr when they are done with it, but this doesn't work because the session objects are still registered as the exchange delegates for their exchanges.

This breaks object lifetime because it then becomes impossible to destroy or reuse the session objects, as it cannot be known when their exchange contexts will be closed.  The result can be use-after-free if the session objects are destroyed, or callback to unrelated session establishments if the `PASESession` or `CASESession` objects are reused (as for instance `CASEClient` does).

 #### Proposed Solution

Replace all instances of `mExchangeCtxt = nullptr;` in `PASESession` and `CASESession` with `CloseExchange`.

Alternatively, if it really is necessary to set to nullptr and not close (I don't know why this would be the case), the session objects must at least `SetDelegate(nullptr)` on `mExchangeCtxt` before clearing their exchange context references.

msandstedt, 2021-12-19:
CC @pan-apple , @bzbarsky-apple

bzbarsky-apple, 2021-12-20:
Looking at the places that null out `mExchangeCtxt` in `CASESession`:
* `CASESession::OnResponseTimeout`: explicitly documents that we are nulling it out because the exchange will be closing itself, so we want to avoid a double-close.
* `CASESession::HandleSigma2Resume`: Does not send a response or claim it will, so the exchange will close itself as soon as this function returns.  We could _probably_ not null things out here as long as we ensured that we null the exchange pointer out in `OnExchangeClosing` (which we would need to implement).
* `CASESession::HandleSigma3`: Same thing as Sigma2Resume.
* `CASESession::OnSuccessStatusReport`: Same thing as Sigma2Resume.
* `CASESession::OnMessageReceived`: explicitly documents that we are nulling it out because the exchange will be closing itself, so we want to avoid a double-close.

Similar for `PASESession`:
* `PASESession::OnResponseTimeout`: explicitly documents that we are nulling it out because the exchange will be closing itself, so we want to avoid a double-close.
* `PASESession::HandleMsg3`: Does not send a response or claim it will, so the exchange will close itself as soon as this function returns.  We could _probably_ not null things out here as long as we ensured that we null the exchange pointer out in `OnExchangeClosing` (which we would need to implement).
* `PASESession::OnSuccessStatusReport`: Same thing as HandleMsg3.
* `PASESession::OnMessageReceived`: explicitly documents that we are nulling it out because the exchange will be closing itself, so we want to avoid a double-close.

msandstedt, 2021-12-22:
@bzbarsky-apple ,

Thanks for the response.  This is slightly difficult then:

```
void CASESession::OnSuccessStatusReport()
{
    ChipLogProgress(SecureChannel, "Success status report received. Session was established");
    mCASESessionEstablished = true;

    // Forget our exchange, as no additional messages are expected from the peer
```

This is not in fact what's happening.  We are getting more messages after this.  I'm not sure how, but I don't know that that matters.

Anyway, the code wants to ACK these.  We can't un-register ourselves as delegate because then SendMessage  for the attempted ACK calls a pure vitrtual method.  Oops:

```
Program terminated with signal SIGABRT, Aborted.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
49	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
[Current thread is 1 (Thread 0x7f0902ffd640 (LWP 3259311))]
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
#1  0x00007f09095ec864 in __GI_abort () at abort.c:79
#2  0x00007f0909850911 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007f090985c38c in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007f090985c3f7 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007f090985d155 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x000055d9d4c41c40 in chip::Messaging::ExchangeMessageDispatch::SendMessage(chip::SessionHandle const&, unsigned short, bool, chip::Messaging::ReliableMessageContext*, bool, chip::Protocols::Id, unsigned char, chip::System::PacketBufferHandle&&)
    (this=0x7f08f000c2a8, session=..., exchangeId=25671, isInitiator=true, reliableMessageContext=0x55d9d59c07c0, isReliableTransmission=false, protocol=..., type=16 '\020', message=...) at ../src/messaging/ExchangeMessageDispatch.cpp:48
#7  0x000055d9d4c40188 in chip::Messaging::ExchangeContext::SendMessage(chip::Protocols::Id, unsigned char, chip::System::PacketBufferHandle&&, chip::BitFlags<chip::Messaging::SendMessageFlags, unsigned short> const&)
```

And we can't free ourselves because we dan't unregister ourselves.  So I think it may only be safe to free (or reuse) CASESession after we get the ExchangeContext::OnExchangeClosing callback, which neither CASESession nor PASESession currently register for.

msandstedt, 2021-12-23:
@bzbarsky-apple , 

I moved discussion of the SIGABRT to https://github.com/project-chip/connectedhomeip/issues/13237.

Here's the fix I currently have for this issue, #13146.  It works.  I will incorporate this into a PR soon.

```
    Fix PASESession and CASESession use-after-free
    
    PASESession and CASESession clear exchange contexts for which they are
    registered as delegate without closing them.  This leads to a situation
    where exchanges can callback into these objects after they are freed.
    This commit fixes the problem by setting the exchange context delegates
    to nullptr before clearing the exchange contexts.
    
    Fixes #13146

diff --git a/src/protocols/secure_channel/CASESession.cpp b/src/protocols/secure_channel/CASESession.cpp
index 8c71b193a..f32d668c7 100644
--- a/src/protocols/secure_channel/CASESession.cpp
+++ b/src/protocols/secure_channel/CASESession.cpp
@@ -255,6 +255,7 @@ void CASESession::OnResponseTimeout(ExchangeContext * ec)
     mDelegate->OnSessionEstablishmentError(CHIP_ERROR_TIMEOUT);
     // Null out mExchangeCtxt so that Clear() doesn't try closing it.  The
     // exchange will handle that.
+    mExchangeCtxt->SetDelegate(nullptr);
     mExchangeCtxt = nullptr;
     Clear();
 }
@@ -684,6 +685,7 @@ CHIP_ERROR CASESession::HandleSigma2Resume(System::PacketBufferHandle && msg)
     mCASESessionEstablished = true;
 
     // Forget our exchange, as no additional messages are expected from the peer
+    mExchangeCtxt->SetDelegate(nullptr);
     mExchangeCtxt = nullptr;
 
     // Call delegate to indicate session establishment is successful
@@ -1118,6 +1120,7 @@ CHIP_ERROR CASESession::HandleSigma3(System::PacketBufferHandle && msg)
     mCASESessionEstablished = true;
 
     // Forget our exchange, as no additional messages are expected from the peer
+    mExchangeCtxt->SetDelegate(nullptr);
     mExchangeCtxt = nullptr;
 
     // Call delegate to indicate session establishment is successful
@@ -1302,6 +1305,7 @@ void CASESession::OnSuccessStatusReport()
     mCASESessionEstablished = true;
 
     // Forget our exchange, as no additional messages are expected from the peer
+    mExchangeCtxt->SetDelegate(nullptr);
     mExchangeCtxt = nullptr;
 
     // Call delegate to indicate pairing completion
@@ -1524,6 +1528,7 @@ exit:
     {
         // Null out mExchangeCtxt so that Clear() doesn't try closing it.  The
         // exchange will handle that.
+        mExchangeCtxt->SetDelegate(nullptr);
         mExchangeCtxt = nullptr;
         Clear();
         mDelegate->OnSessionEstablishmentError(err);
diff --git a/src/protocols/secure_channel/PASESession.cpp b/src/protocols/secure_channel/PASESession.cpp
index aadda0f8f..52c7e9f1b 100644
--- a/src/protocols/secure_channel/PASESession.cpp
+++ b/src/protocols/secure_channel/PASESession.cpp
@@ -346,6 +346,7 @@ void PASESession::OnResponseTimeout(ExchangeContext * ec)
     mDelegate->OnSessionEstablishmentError(CHIP_ERROR_TIMEOUT);
     // Null out mExchangeCtxt so that Clear() doesn't try closing it.  The
     // exchange will handle that.
+    mExchangeCtxt->SetDelegate(nullptr);
     mExchangeCtxt = nullptr;
     Clear();
 }
@@ -824,6 +825,7 @@ CHIP_ERROR PASESession::HandleMsg3(System::PacketBufferHandle && msg)
     mPairingComplete = true;
 
     // Forget our exchange, as no additional messages are expected from the peer
+    mExchangeCtxt->SetDelegate(nullptr);
     mExchangeCtxt = nullptr;
 
     // Call delegate to indicate pairing completion
@@ -843,6 +845,7 @@ void PASESession::OnSuccessStatusReport()
     mPairingComplete = true;
 
     // Forget our exchange, as no additional messages are expected from the peer
+    mExchangeCtxt->SetDelegate(nullptr);
     mExchangeCtxt = nullptr;
 
     // Call delegate to indicate pairing completion
@@ -938,6 +941,7 @@ exit:
     {
         // Null out mExchangeCtxt so that Clear() doesn't try closing it.  The
         // exchange will handle that.
+        mExchangeCtxt->SetDelegate(nullptr);
         mExchangeCtxt = nullptr;
         Clear();
         ChipLogError(SecureChannel, "Failed during PASE session setup. %s", ErrorStr(err));
```

msandstedt, 2021-12-23:
Apparently this is known:

```
void OperationalDeviceProxy::DeferCloseCASESession()
{
    // Defer the release for the pending Ack to be sent
    mSystemLayer->ScheduleWork(CloseCASESessionTask, this);
}
```

The above does work.

My personal critique is that it would be very useful if success callbacks could actually indicate that the stack is done with an object.  For this problem (and #13237), this would imply sending ACK and closing the exchange _before_ calling back.  That would be far easier to deal with.  This all looks really fragile.

msandstedt, 2022-01-07:
#12794 allows us to have a real fix for this where we no longer need to defer reuse of the CASESession object (and equivalently, the PASESession object) until after the next iteration of the event loop.

I will open a PR to make that change.

closed 2022-01-07 by msandstedt