project-chip/connectedhomeip #13163: Java clients should switch to managing ReadClient objects directly for subscriptions

mrjerryjohns, 2021-12-20:
#### Problem

Java clients today rely on the 'fire-and-forget' methods in CHIPCluster.h to setup a subscription to a particular node, and then calling `ShutdownSubscriptions` and `ShutdownSubscription` methods on the IM Engine to tear down a particular subscription, or all subscriptions to a particular node.

This subverts the ownership model, where while the application should be truly responsible for the lifetime of the subscription object, but instead, the ownership is managed by the stack instead.

It also avoids problems with leaked ReadClients because there is a strong ownership model that can be put in place using smart pointers.

mrjerryjohns, 2021-12-20:
FYI @austinh0

woody-apple, 2022-01-21:
Hard to tell, is this complete in the linked PR? @mrjerryjohns

closed 2022-01-21 by woody-apple