project-chip/connectedhomeip #13165: Error in ./scripts/tools/zap/generate.py : No such file or directory: 'node'

simonseres, 2021-12-20:
#### Problem
<what's wrong or missing, please include any applicable:
Followed the [guide to build](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md) on an Ubuntu machine. 

Environment check was successful: `Environment looks good, you are ready to go!` though after starting the build I run into 
```Traceback (most recent call last):
  File "/home/dev/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 165, in <module>
    main()
  File "/home/dev/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 153, in main
    runGeneration(zap_file, zcl_file, templates_file, output_dir)
  File "/home/dev/connectedhomeip/scripts/examples/../../scripts/tools/zap/generate.py", line 99, in runGeneration
    subprocess.check_call(['node', './src-script/zap-generate.js', '-z',
  File "/usr/lib/python3.9/subprocess.py", line 368, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.9/subprocess.py", line 349, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'node'
```
Tried to understand the code, but failed so far - some help is appreciated.

simonseres, 2021-12-20:
Ok - I found out that some packages are missing and installed missing libraries / software. Now I am stuck at
```
None
Traceback (most recent call last):
  File "/home/dev/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 679, in <module>
    sys.exit(main())
  File "/home/dev/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 671, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/dev/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 414, in setup
    result = step(spin)
  File "/home/dev/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 533, in virtualenv
    if not virtualenv_setup.install(
  File "/home/dev/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 195, in install
    _check_call(cmd, env=envcopy)
  File "/home/dev/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 87, in _check_call
    subprocess.check_call(args, **kwargs)
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/dev/connectedhomeip/.environment/pigweed-venv/bin/python3', '-m', 'venv', '--upgrade', '/home/dev/connectedhomeip/.environment/pigweed-venv']' returned non-zero exit status 1.
```

simonseres, 2021-12-20:
got that one - just some missing dependencies. (But why didn't the script take care of that?)

closed 2021-12-20 by simonseres

RandolphChina, 2022-05-26:
> got that one - just some missing dependencies. (But why didn't the script take care of that?)

Dear friend, I meet the same problem , how to solove it  please?