project-chip/connectedhomeip #13168: Spec-required level control interactions were removed from on/off cluster

bzbarsky-apple, 2021-12-20:
#### Problem
https://github.com/project-chip/connectedhomeip/pull/12507 removed some level control bits from the on/off cluster, claiming they are not spec compliant.  But as far as I can tell they are in fact required by the spec.

The requirement is a COMEFROM-style requirement in the _level control_ cluster, so it's not obvious from reading just the on/off cluster.  The relevant spec section is "1.6.4.1.1. Effect of On/Off Commands on the CurrentLevel Attribute" in the appclusters spec.  The PR says no automated test regressions were caused, but all that says to me is that we have no test coverage for this part of the spec and should add some.

 #### Proposed Solution
Double-check what implements this spec section in our code, write tests for it, make sure we pass those tests.

bzbarsky-apple, 2021-12-20:
@mkardous-silabs

bzbarsky-apple, 2021-12-20:
@jmartinez-silabs @Damian-Nordic

mkardous-silabs, 2021-12-20:
@bzbarsky-apple Fix is on it's way with an update on the level control and the OnOff cluster.

PR #13338 (2022-01-14): Update/Add Level control features based on lastest cluster revision
#### Problem
Some level control lighting behaviours were disable by ember and zll non existant defines

#### Change overview
- Add runtime attribute metadata checks to validate that the needed attributes exist for that feature. Replace the #ifdef gates to compile that code by default, but still allow toggling it off by defining the right IGNORE_LEVEL_CONTROL_XXX defines
- Enable Feature map for level control and update cluster revision
- Fix min and max Level usage considering enabled attributes and level control feature map
- [... cut]
closed 2022-01-14 by jmartinez-silabs