project-chip/connectedhomeip #13180: CHIP Tool - Missing Attribute- NOC's for Node Operational Credentials Cluster

sumaky, 2021-12-21:
#### Problem

 While using CHIP tool when we execute the following command the below attributes gets listed 

**./chip-tool operationalcredentials read attribute-name [param1 param2 ...]**

 +-------------------------------------------------------------------------------------+                                               
  | Attributes:                                                                                                                   
  +-------------------------------------------------------------------------------------+                                               
  | * fabrics-list                                                                                                               
  | * supported-fabrics                                                                                                              
  | * commissioned-fabrics                                                                                                        
  | * trusted-root-certificates                                                                                                 
  | * current-fabric-index                                                                                                           
  | * attribute-list                                                                                                                
  | * cluster-revision                                                                                                              
  +-------------------------------------------------------------------------------------+ 
Note that NOC's Attribute is missing in the list

Spec Reference:11.18.6. Attributes

 #### Proposed Solution
Add the NOC's attribute to the list. This is used to do the verification steps mapping to the test plan.

bzbarsky-apple, 2022-05-05:
Here's what I see on tip (b963a3fcd593631b945edddb7f8e9e4679f1555d)
```
  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * nocs                                                                              |
  | * fabrics                                                                           |
  | * supported-fabrics                                                                 |
  | * commissioned-fabrics                                                              |
  | * trusted-root-certificates                                                         |
  | * current-fabric-index                                                              |
  | * generated-command-list                                                            |
  | * accepted-command-list                                                             |
  | * attribute-list                                                                    |
  | * feature-map                                                                       |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+
```

closed 2022-05-05 by bzbarsky-apple

sumaky, 2022-05-05:
NOCs: 1 entries
[1650457397.446011][4559:4564] CHIP:TOO:     [1]: {
[1650457397.446083][4559:4564] CHIP:TOO:       Noc: Elided value too large of size 241
[1650457397.446147][4559:4564] CHIP:TOO:       Icac: Elided value too large of size 231
[1650457397.446206][4559:4564] CHIP:TOO:       FabricIndex: 1
[1650457397.446262][4559:4564] CHIP:TOO:      }

sumaky, 2022-05-05:
@bzbarsky-apple Though NOC is listed we are not able to see the values . I guess this is expected behavior. We may not be able to do the verification steps for the Node Operational Test Cases.

bzbarsky-apple, 2022-05-05:
We could try to do something to log those values if it would be useful.  Please file issues as needed.