project-chip/connectedhomeip #13189: Android app build fails in t26.lts branch

manimathma, 2021-12-21:
Dear team, 

I am consistently getting following error why trying to build android chiptool app. 

> ...connectedhomeip/out/android_arm64/CMakeLists.txt : C/C++ debug|arm64-v8a : CMake Error at ....connectedhomeip/out/android_arm64/CMakeLists.txt:258 (add_custom_target):
    Cannot find source file:

gradle.properties:
`# Project-wide Gradle settings.
# IDE (e.g. Android Studio) users:
# Gradle settings configured through the IDE *will override*
# any settings specified in this file.
# For more details on how to configure your build environment visit
# http://www.gradle.org/docs/current/userguide/build_environment.html
# Specifies the JVM arguments used for the daemon process.
# The setting is particularly useful for tweaking memory settings.
org.gradle.jvmargs=-Xmx2048m
# When configured, Gradle will run in incubating parallel mode.
# This option should only be used with decoupled projects. More details, visit
# http://www.gradle.org/docs/current/userguide/multi_project_builds.html#sec:decoupled_projects
# org.gradle.parallel=true
# AndroidX package structure to make it clearer which packages are bundled with the
# Android operating system, and which are packaged with your app"s APK
# https://developer.android.com/topic/libraries/support-library/androidx-rn
android.useAndroidX=true
# Automatically convert third-party libraries to use AndroidX
android.enableJetifier=true
# Point to the SDK build dir without quotes (../../../../out/android_arm64 for
# example) to build SDK from source code and debug in Android Studio.
# Set to blank to use the SDK prebuilt by scripts/build/build_examples.py.
matterBuildSrcDir=../../../../out/android_arm64`

Few lines before 258:
`#//build/chip/java:shared_cpplib
set("target" "build__chip__java_shared_cpplib")
set("${target}__other_srcs" "/Applications/Android Studio.app/Contents/plugins/android-ndk/sources/cxx-stl/llvm-libc++/libs/arm64-v8a/libc++_shared.so")
set("${target}__output" "/Users/amanimu/Documents/pecan/ACS-CHIP/connectedhomeip/out/android_arm64/lib/jni/arm64-v8a/libc++_shared.so")
add_custom_command(OUTPUT ${${target}__output}
  COMMAND ${CMAKE_COMMAND} -E copy "/Applications/Android Studio.app/Contents/plugins/android-ndk/sources/cxx-stl/llvm-libc++/libs/arm64-v8a/libc++_shared.so" "/Users/amanimu/Documents/pecan/ACS-CHIP/connectedhomeip/out/android_arm64/lib/jni/arm64-v8a/libc++_shared.so"
  DEPENDS  ${${target}__other_srcs}
  WORKING_DIRECTORY "/Users/amanimu/Documents/pecan/ACS-CHIP/connectedhomeip/out/android_arm64/"
  COMMENT "Copy ${target}"
  VERBATIM)
add_custom_target("${target}" SOURCES ${${target}__other_srcs} DEPENDS ${${target}__output})`

manimathma, 2021-12-21:
Seems like i didn't set the NDK path properly. Validating by setting NDK path properly.
After following this https://stackoverflow.com/questions/64372383/ndk-at-library-android-sdk-ndk-bundle-did-not-have-a-source-properties-file issue  this issue is not observed.

closed 2021-12-21 by manimathma

mohitjadav86, 2022-08-31:
hi @manimathma  i'm facing this issue https://github.com/project-chip/connectedhomeip/issues/22256 can you help on this ?