project-chip/connectedhomeip #13214: Update commands media clusters

lazarkov, 2021-12-22:
#### Problem
- LaunchAppRequest data argument is not OCTET_STRING as per spec.

 #### Proposed Solution
- Update per spec

turon, 2022-02-01:
Media clusters must comply with spec for v1.
Removing tag `v1_triage_split_4`

lazarkov, 2022-02-08:
- ~~LaunchApp Request data argument is not OCTET_STRING as per spec.~~ [url] (https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/application-launcher-cluster.xml#L31-L51)
- ~~Launcher Response data argument is not OCTET_STRING as per spec.~~ [url] (https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/application-launcher-cluster.xml#L31-L51)
- ~~Media Playback start time should be EPOCH_US as per spec.~~ [url](https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/media-playback-cluster.xml#L30)
- Media Playback commands should be optional as per spec. (https://github.com/project-chip/connectedhomeip/pull/13195#discussion_r773542618)
- Target Navigator command should be optional as per spec. (https://github.com/project-chip/connectedhomeip/pull/13195#discussion_r773544026)
- Target Navigator argument should be optional as per spec. (https://github.com/project-chip/connectedhomeip/pull/13195#discussion_r773543881)
- Application Launcher, Data Type Application.Endpoint should be endpoint-no as per spec. (https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/media/ApplicationLauncher.adoc#application-3)

closed 2022-02-09 by lazarkov