project-chip/connectedhomeip #13223: ESP32 Comissionion over ble after restart device

srdantas, 2021-12-22:
Problem: Following the README from the temperature-app to ESP32 example I am trying to commission the device through ble but for some reason the device cannot be found by the controller. The first time I ran this process it worked perfectly, but after restarting the device, I can no longer commission it or connect to it. I reset the device several times to clear a possible storage but it had no effect. The scenario takes place using the python controller or chip-tool. Successful for commissioning done over IP when the device had the local network credentials configured.

 * expected behavior: connect to comissioneble device, or comissionion it
 * actual behavior: canot connect to device and canot comissionion over ble
 * steps to reproduce: run the readme from example and restart device (I does not execute close-ble and dont remeber the assigment nodeId in that case)
 
#### System configuration
 * Controller: Run and build in MacOS 12.1 Intel
 * Device: ESP32-DEVKitC-v4 with ESP-IDF v4.4

#### Logs:
[chip-device-ctrl_log.txt](https://github.com/project-chip/connectedhomeip/files/7764571/chip-device-ctrl_log.txt)
[device_log.txt](https://github.com/project-chip/connectedhomeip/files/7764572/device_log.txt)

 #### Proposed Solution
I can't propose a solution because I don't have access to the specification and so I don't know the expected behavior here, when someone proposes a solution I can fix it myself.

dhrishi, 2021-12-22:
@srdantas If you are using Python controller (chip-device-ctrl) then it is not supported yet. To retry, you can erase flash on ESP32 (idf.py erase_flash) so that you will be able to commission it again.

On a different note, you can instead use https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool for commissioning. You can control the device (operational communication) even after it is rebooted after commissioning is done.

srdantas, 2021-12-22:
@dhrishi i try with chip-too, before run erase_flash in device, and got this log from chip-tool:
```
❯ ./chip-tool pairing ble-wifi 248163 2.4_2906 my_pass 20202021 3840
[1640208814885] [8110:57801] CHIP: [DL] _Init
[1640208814885] [8110:57801] CHIP: [IN] UDP::Init bind&listen port=5542
[1640208814885] [8110:57801] CHIP: [IN] UDP::Init bound to port=5542
[1640208814885] [8110:57801] CHIP: [IN] UDP::Init bind&listen port=5542
[1640208814885] [8110:57801] CHIP: [IN] UDP::Init bound to port=5542
[1640208814885] [8110:57801] CHIP: [IN] TransportMgr initialized
[1640208814886] [8110:57801] CHIP: [DIS] Init fabric pairing table with server storage
[1640208814886] [8110:57801] CHIP: [ZCL] Using ZAP configuration...
[1640208814886] [8110:57801] CHIP: [CTL] System State Initialized...
[1640208814886] [8110:57801] CHIP: [CTL] Generating NOC
[1640208814887] [8110:57801] CHIP: [CTL] Generating ICAC
[1640208814887] [8110:57801] CHIP: [DIS] Verifying the received credentials
[1640208814887] [8110:57801] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1640208814887] [8110:57801] CHIP: [IN] 0x04, 0xf1, 0xf7, 0x8d, 0x14, 0x4b, 0xfa, 0xca,
[1640208814887] [8110:57801] CHIP: [IN] 0xe5, 0xab, 0xfc, 0x63, 0x69, 0x38, 0x14, 0x43,
[1640208814887] [8110:57801] CHIP: [IN] 0x23, 0xba, 0xaa, 0x6d, 0x92, 0xf6, 0xe4, 0x82,
[1640208814887] [8110:57801] CHIP: [IN] 0x92, 0xdc, 0x8e, 0x9c, 0x6a, 0x7c, 0x3b, 0xd0,
[1640208814887] [8110:57801] CHIP: [IN] 0xf1, 0x41, 0x1c, 0x2c, 0x70, 0x94, 0x92, 0x11,
[1640208814887] [8110:57801] CHIP: [IN] 0xa1, 0x8e, 0xba, 0xb2, 0x44, 0x49, 0x18, 0x87,
[1640208814887] [8110:57801] CHIP: [IN] 0x18, 0xac, 0xfa, 0xfe, 0xa9, 0xd1, 0xa4, 0xa9,
[1640208814888] [8110:57801] CHIP: [IN] 0xc8, 0x23, 0x2e, 0x6d, 0x2e, 0xbf, 0xa6, 0x40,
[1640208814888] [8110:57801] CHIP: [IN] 0x74,
[1640208814888] [8110:57801] CHIP: [IN] Generated compressed fabric ID
[1640208814888] [8110:57801] CHIP: [IN] 0xc2, 0x1f, 0xf5, 0x3b, 0x94, 0x3f, 0x16, 0xc8,
[1640208814888] [8110:57801] CHIP: [DIS] Added new fabric at index: 1, Initialized: 1
[1640208814888] [8110:57801] CHIP: [DIS] Assigned compressed fabric ID: 0xC21FF53B943F16C8, node ID: 0x000000000001B669
[1640208814888] [8110:57801] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0xC21FF53B943F16C8
[1640208814888] [8110:57801] CHIP: [IN] UDP::Init bind&listen port=5550
[1640208814888] [8110:57801] CHIP: [IN] UDP::Init bound to port=5550
[1640208814888] [8110:57801] CHIP: [IN] UDP::Init bind&listen port=5550
[1640208814888] [8110:57801] CHIP: [IN] UDP::Init bound to port=5550
[1640208814888] [8110:57801] CHIP: [IN] TransportMgr initialized
[1640208814888] [8110:57801] CHIP: [CTL] Generating NOC
[1640208814888] [8110:57801] CHIP: [CTL] Generating ICAC
[1640208814888] [8110:57801] CHIP: [DIS] Verifying the received credentials
[1640208814889] [8110:57801] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1640208814889] [8110:57801] CHIP: [IN] 0x04, 0xf1, 0xf7, 0x8d, 0x14, 0x4b, 0xfa, 0xca,
[1640208814889] [8110:57801] CHIP: [IN] 0xe5, 0xab, 0xfc, 0x63, 0x69, 0x38, 0x14, 0x43,
[1640208814889] [8110:57801] CHIP: [IN] 0x23, 0xba, 0xaa, 0x6d, 0x92, 0xf6, 0xe4, 0x82,
[1640208814889] [8110:57801] CHIP: [IN] 0x92, 0xdc, 0x8e, 0x9c, 0x6a, 0x7c, 0x3b, 0xd0,
[1640208814889] [8110:57801] CHIP: [IN] 0xf1, 0x41, 0x1c, 0x2c, 0x70, 0x94, 0x92, 0x11,
[1640208814889] [8110:57801] CHIP: [IN] 0xa1, 0x8e, 0xba, 0xb2, 0x44, 0x49, 0x18, 0x87,
[1640208814889] [8110:57801] CHIP: [IN] 0x18, 0xac, 0xfa, 0xfe, 0xa9, 0xd1, 0xa4, 0xa9,
[1640208814889] [8110:57801] CHIP: [IN] 0xc8, 0x23, 0x2e, 0x6d, 0x2e, 0xbf, 0xa6, 0x40,
[1640208814889] [8110:57801] CHIP: [IN] 0x74,
[1640208814889] [8110:57801] CHIP: [IN] Generated compressed fabric ID
[1640208814889] [8110:57801] CHIP: [IN] 0x56, 0x8c, 0x76, 0xa1, 0x08, 0xe9, 0x51, 0x0b,
[1640208814889] [8110:57801] CHIP: [DIS] Added new fabric at index: 2, Initialized: 1
[1640208814889] [8110:57801] CHIP: [DIS] Assigned compressed fabric ID: 0x568C76A108E9510B, node ID: 0x000000000001B669
[1640208814889] [8110:57801] CHIP: [CTL] Joined the fabric at index 2. Compressed fabric ID is: 0x568C76A108E9510B
[1640208814889] [8110:57801] CHIP: [IN] UDP::Init bind&listen port=5550
[1640208814889] [8110:57801] CHIP: [IN] UDP::Init bound to port=5550
[1640208814889] [8110:57801] CHIP: [IN] UDP::Init bind&listen port=5550
[1640208814889] [8110:57801] CHIP: [IN] UDP::Init bound to port=5550
[1640208814889] [8110:57801] CHIP: [IN] TransportMgr initialized
[1640208814890] [8110:57801] CHIP: [CTL] Generating NOC
[1640208814890] [8110:57801] CHIP: [CTL] Generating ICAC
[1640208814890] [8110:57801] CHIP: [DIS] Verifying the received credentials
[1640208814890] [8110:57801] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000003 and root pubkey
[1640208814890] [8110:57801] CHIP: [IN] 0x04, 0xf1, 0xf7, 0x8d, 0x14, 0x4b, 0xfa, 0xca,
[1640208814890] [8110:57801] CHIP: [IN] 0xe5, 0xab, 0xfc, 0x63, 0x69, 0x38, 0x14, 0x43,
[1640208814890] [8110:57801] CHIP: [IN] 0x23, 0xba, 0xaa, 0x6d, 0x92, 0xf6, 0xe4, 0x82,
[1640208814890] [8110:57801] CHIP: [IN] 0x92, 0xdc, 0x8e, 0x9c, 0x6a, 0x7c, 0x3b, 0xd0,
[1640208814890] [8110:57801] CHIP: [IN] 0xf1, 0x41, 0x1c, 0x2c, 0x70, 0x94, 0x92, 0x11,
[1640208814890] [8110:57801] CHIP: [IN] 0xa1, 0x8e, 0xba, 0xb2, 0x44, 0x49, 0x18, 0x87,
[... cut ...]
[1640208854895] [8110:57801] CHIP: [TOO] Run command failure: examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1536: CHIP Error 0x00000032: Timeout
[1640208854900] [8110:57801] CHIP: [SPT] VerifyOrDie failure at examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
[1]    8110 abort      ./chip-tool pairing ble-wifi 248163 2.4_2906 my_pass 20202021 3840
```

And in device log, are equals at [device_log.txt](https://github.com/project-chip/connectedhomeip/files/7764572/device_log.txt)

dhrishi, 2021-12-23:
This is strange. Can you please share the GIT commit head that you are using?

srdantas, 2021-12-23:
In project-chip commit id is feb07d5cd and in idf a79dc75f0a. I recently updated same projects.

closed 2022-10-18 by stale[bot]