project-chip/connectedhomeip #13227: ResolverDelegateProxy use-after-free

msandstedt, 2021-12-22:
#### Problem

Discovery_ImplPlatform.cpp decrements the reference counter for the ResolverDelegateProxy by calling Release, which eventually causes it to be freed.  However, this doesn't necessarily un-register the freed delegate from e.g. Resolver_ImplMinimalMdns.cpp.  The result is that if more records are discovered than expected, code like Resolver_ImplMinimalMdns.cpp can execute methods on the freed ResolverDelegateProxy.

 #### Proposed Solution

Something like this might work:

```
diff --git a/src/lib/dnssd/ResolverProxy.h b/src/lib/dnssd/ResolverProxy.h
index 5d3683387..3484366a9 100644
--- a/src/lib/dnssd/ResolverProxy.h
+++ b/src/lib/dnssd/ResolverProxy.h
@@ -28,6 +28,12 @@ class ResolverDelegateProxy : public ReferenceCounted<ResolverDelegateProxy>, pu
 public:
     void SetDelegate(ResolverDelegate * delegate) { mDelegate = delegate; }
 
+    void Release()
+    {
+        chip::Dnssd::Resolver::Instance().SetResolverDelegate(nullptr);
+        ReferenceCounted<ResolverDelegateProxy>::Release();
+    }
+
     /// ResolverDelegate interface
     void OnNodeIdResolved(const ResolvedNodeData & nodeData) override
     {
```

msandstedt, 2021-12-22:
CC @vivien-apple

msandstedt, 2021-12-22:
The suggested fix does fix my problem.  However, now things like TestCommissionManager are failing.

I don't really know what should be done here.  If the resolver delegate proxy frees itself, it also needs to unregister itself.  But this seems to be breaking other things.

msandstedt, 2021-12-23:
Well, I don't know what the real fix is.

For now, I am calling this in my code after Shutdown:

```
chip::Dnssd::Resolver::Instance().SetResolverDelegate(nullptr);
```

msandstedt, 2021-12-29:
Confirmed that the same problem can also occur in Darwin.  I was not able to overcome this without Retaining and effectively leaking the ResolverDelegateProxy when shutting down ResolverProxy.

msandstedt, 2021-12-30:
Here is one bug at least:

```
static void HandleNodeBrowse(void * context, DnssdService * services, size_t servicesSize, CHIP_ERROR error)
{
    ResolverDelegateProxy * proxy = static_cast<ResolverDelegateProxy *>(context);
    proxy->Release();

    for (size_t i = 0; i < servicesSize; ++i)
    {
        proxy->Retain();
        // For some platforms browsed services are already resolved, so verify if resolve is really needed or call resolve callback
        if (!services[i].mAddress.HasValue())
        {
            ChipDnssdResolve(&services[i], services[i].mInterface, HandleNodeResolve, context);
        }
        else
        {
            HandleNodeResolve(context, &services[i], error);
        }
    }
}
```

Coming into this method, I have reference count of 1, which immediately decrements to 0.  Then we increment back to 1, which is use-after-free.  This method either needs to have retain moved to the end, or must not call retain at all.

bzbarsky-apple, 2022-06-13:
Shouldn't ~ResolverDelegateProxy be handling the un-registration?

Removing "deferral candidate" label, since this is a use-after-free bug, hence likely exploitable.

andy31415, 2022-09-28:
@msandstedt Trying to understand if this is still an issue and to understand it in general: this mentiones MinimalMdns but also Discovery_ImplPlatform.cpp ... those two should be mutually exclusive.

Is this still an issue? I am unsure how to reproduce. MinMDNS code was updated quite a bit with the usage of address resolver since last year.

andy31415, 2022-09-28:
MinMDNS does not currently use ResolverProxy at all.

andy31415, 2022-09-29:
Closing: based on discussion with @msandstedt since MinMDNS is not using resolver proxy, this should be fine now.

closed 2022-09-29 by andy31415