project-chip/connectedhomeip #13231: Armv5 does not support atomic operation natively

bekencorp, 2021-12-23:
#### Problem
I'm using armv5te/freertos to porting this SDK, when i compiled , i encounted some errors about the buit-in atomic function
arm-none-eabi/include/c++/10.3.1/bits/atomic_base.h:548: undefined reference to `__atomic_fetch_add_4'
arm-none-eabi/include/c++/10.3.1/bits/atomic_base.h:334: undefined reference to `__atomic_fetch_sub_4'
arm-none-eabi/include/c++/10.3.1/bits/atomic_base.h:509: undefined reference to `__atomic_compare_exchange_4'
and i tried to link the atomic library, but can't find it.

 #### Proposed Solution
I have no idea how to resovle this issue. i know armv5te is very old.
Can we have toggle to disable the atomic ?

rpavlik, 2021-12-28:
I wonder if you can substitute a disable/restore interrupts pair as part of the atomic implementation. I think that might be enough to satisfy the requirements? (If interrupts are disabled, and no other task can start running, it's as if the write was atomic, I think)

rpavlik, 2022-01-02:
Actually I think the best way to go is to use the mit-licensed sources from libatomic-ops to substitute for the missing operations. That library does appear like it should work with the one atomic swap primitive in arm v5

jamesluo11, 2022-01-26:
Yeah, @rpavlik Do you mean this library https://packages.debian.org/source/sid/libatomic-ops ?

rpavlik, 2022-01-26:
Yes, that's the one

closed 2022-09-22 by stale[bot]