project-chip/connectedhomeip #13268: Where can found Matter specification ?

standback, 2021-12-29:
#### Problem
Hello guys.

Does anyone know where I can find the matter specification that can describe every detail about  Matter? 

for now, I only know this git repo, which is an official open-source implement of Matter.

thanks.

jonsmirl, 2021-12-31:
You have to pay $20,000 to be able to see it.

standback, 2022-01-04:
got it.
thanks for your reply.

thanks a lot.

shishkin, 2022-01-05:
> You have to pay $20,000 to be able to see it.

Thanks @jonsmirl. Could you please provide a reference to the terms of that deal?

jonsmirl, 2022-01-05:
https://zigbeealliance.org/join/

You have to pay the $20K. At $7K they give you a useless three month old
spec.

jonsmirl, 2022-01-05:
Matter is just Zigbee warmed over, you can get 80% of what you need to know from this:
https://www.nxp.com/docs/en/user-guide/JN-UG-3115.pdf

And then look at the example apps to get the rest.

closed 2022-01-06 by andy31415