project-chip/connectedhomeip #13296: Does ios 15.2 (beta) home App support Matter devices?

nicolabortignon, 2022-01-01:
I believe Apple have started supporting Matter in their ios 15. 
I was wondering, if any of you is aware if the 15.2 beta version of Home (the ios app that integrates homekit) can support Matter. 

I tried to use the QR-code generated by an ESP32 device that is currently loading this repo latest version, in the 'adding device' onboarding screen of home, but without success.

Raqbit, 2022-01-03:
I believe the Matter support in Home requires a device profile to function, see the instructions here (found on https://developer.apple.com/documentation/homekit):

https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableMatter_Instructions.pdf

oleksiikutuzov, 2022-01-03:
The thing is this paper only mentions 15.1 and wasn't updated since then. Also the profile expired on January 1st and can't be installed anymore

nicolabortignon, 2022-01-03:
Thank you @Raqbit. As @oleksiikutuzov pointed out this has now expired. 
Also, would you mind confirm that a Apple Developer account (and subscription) are needed in order to install that Profile Certificate?
It would not install on my device, but I'm not sure if it is because it has expired or because my apple id dev subscription was not renewed in a while.

Raqbit, 2022-01-10:
> Thank you @Raqbit. As @oleksiikutuzov pointed out this has now expired. Also, would you mind confirm that a Apple Developer account (and subscription) are needed in order to install that Profile [...]

@nicolabortignon It appears the profile did indeed expire itself on my test device, which I expect is also the reason why it won't install for you. It also appears that the 15.3 Home app also does not support Matter without certificate either, as it does not detect Matter QR codes.

nicolabortignon, 2022-01-11:
Thank you for confirming @Raqbit.

How does this group work with Apple on those situations? Is there a private repo where this can be forwarded as a request? Do we have a POC of Apple representative in the CHIP/Matter consortium?

Raqbit, 2022-01-11:
> Thank you for confirming @Raqbit.
> 
> How does this group work with Apple on those situations? Is there a private repo where this can be forwarded as a request? Do we have a POC of Apple representative in the CHIP/Matter consortium?

@nicolabortignon There is an iOS-version of the CHIP-Tool Matter development app, which you can find [here](https://github.com/project-chip/connectedhomeip/tree/master/src/darwin/CHIPTool). For my own testing I've been using the Android version of the same app, which is not super polished but works for me. (though ToT breaks it often)

I'm guessing just like Google's Matter support in their Home app, the Matter support in Apple's Home app will stay unavailable until the final Matter specification & SDK is released somewhere in the second half of this year.

bzbarsky-apple, 2022-01-12:
@sagar-apple and @woody-apple should know more about what's going here.

jakubdybczak, 2022-02-08:
[PDF](https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableMatter_Instructions.pdf) got updated at 29 January and I have successfully installed Matter profile on iOS 15.3.
I also tried launching all-clusters-app on ESP32 and QR code was recognized by Home App!
However, it won't connect with ESP32.

Commit hash suggested in PDF to ensure compatibility (_b717b36130af37479377f236f03db85dd678c645_) is not compatible at all... For example, Apple Home App is sending old handshake frame that was changed in this PR, which was before this suggested commit hash (https://github.com/project-chip/connectedhomeip/pull/10177). I tried changing this handshake frame manually but there are more problems into it. I think that Home App is compatible with some older version of this repository or maybe it will work on iOS 15.4 beta. PDF seems outdated regarding OS version and it is still mentioning iOS 15.1.

It would be nice to make corrections to this document and properly test which repo tag/commit hash is really compatible with Home app. 

@sagar-apple  @woody-apple you guys could probably push this topic forward :)

jakubdybczak, 2022-02-11:
Update: More things are working on iOS 15.4 Beta 2
From UI perspective during paring, dialog window disappears and no device is added. From iPhone logs I can see some error (I attach only few logs):

`DevicePairingDelegate Pairing complete. Status ../../../../../../../../Sources/CHIPFramework/CHIPFramework-38.5.4/connectedhomeip/src/controller/CHIPDeviceController.cpp:1538: Success`
`WiFiNetworkCopyPassword: wifid: Copy password for Network <private>`
`[CHIPAccessoryServer] Sending Thread Credentials...`
`[CHIPAccessoryServer] CHIP Accessory Server failed to send Thread credentials... Error: Error Domain=ThreadCredentialsStore Code=9 "No preferred network found" UserInfo={NSLocalizedDescription=No preferred network found}`
`[CHIPAccessoryServer] Failed to add Network credentials for Type: 0 with Error: Error Domain=CHIPErrorDomain Code=1 "Undefined data model error:1." UserInfo={NSLocalizedDescription=Undefined data model error:1.}`
`[CHIPAccessoryServer] CHIP Accessory pairing failed: Error Domain=CHIPErrorDomain Code=1 "Undefined data model error:1." UserInfo={NSLocalizedDescription=Undefined data model error:1.}`

My ESP32 does not support Thread and it's disabled. Maybe Thread is required for Home App?

ReneJosefsen, 2022-03-31:
Have anyone gotten this to work? I'm trying to setup a light following the provided instructions, it seems to be able to connect to the Thread network but fails just after it has connected and the iPad reports that it is not able to configure the device.

I'm using AppleTV 4K and iPad running 15.4

tewarid, 2022-03-31:
iOS 15.4 Home app works for me with linux examples using the git hash mentioned in latest PDF (link posted above).

jakubdybczak, 2022-03-31:
@tewarid Can you provide information about your setup (which devices are you using)? Is it possible to make it work without Apple TV, or is it required?

ReneJosefsen, 2022-03-31:
As far as I understand, the AppleTV is only required for Thread based devices, since it will act as the border router in the network.

jakubdybczak, 2022-03-31:
@ReneJosefsen I'm asking because on one of WWDC videos (https://developer.apple.com/videos/play/wwdc2021/10298/) at 16:40 timestamp there is a note regarding required Home Hub (so Apple TV / HomePod or iPad I guess).

I have troubles with connecting ESP32 with Matter to HomeKit from iPhone without any Home Hub. It is connecting and properly pairing devices (ESP32 is connecting to my wifi network properly) but after this Home app is telling me that device is unreachable. I wonder if this can be caused by **not** having Home Hub.

tewarid, 2022-03-31:
> @tewarid Can you provide information about your setup (which devices are you using)? Is it possible to make it work without Apple TV, or is it required?

Doesn't work for me without AppleTV. tvOS version is in sync with iOS version. I haven't tried thread but Wi-Fi works with discovery over BLE. Looks like iOS Home app talks to device via hub after initial setup is complete.

plan44, 2022-06-30:
Today the profile to enable matter in iOS mentioned by @Raqbit above expired.

As I was successfully testing `bridge-app` commissioning yesterday, but finding that today the Home.app does not even recognize QR codes any more, only barcodes, I know the profile was relevant and working up to iOS16beta2 😉

Any plans to update the profile in the near future?

wickyou23, 2022-07-05:
Hello @bzbarsky-apple, If iOS devices do not accept the `EnnableMatter` profile in the PDF, does the `CHIP-Tool` app [(here)](https://github.com/project-chip/connectedhomeip/tree/master/src/darwin/CHIPTool) for iOS or macOS still work?

plan44, 2022-07-05:
Note: The profile and the instructions have been updated in the meantime. Just re-download the files again from the links @Raqbit posted above.

But the profile enables matter support built into the OS, CHIP-Tool is completely separate from that and brings its own matter stack along, so does not need the profile (AFAIK, experts please correct me if I'm wrong)

closed 2023-01-14 by stale[bot]

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/