project-chip/connectedhomeip #13303: chip-tool doesn't work on Monterey 15.1 and above

nicolabortignon, 2022-01-03:
Most likely it's a bluetooth issue (as Monterey seems to have few), but wanted to double check if maybe it's something else.

I have two examples here (attempt to commission a node and attempt to send a command to an already commissioned one)

Commissioning a new node 

'''bortignon-mbp:SOME-PATH bortignon$ ./chip-tool pairing ble-wifi 13131 mySSID myPassword 20202021 3840
[1641243251851] [2096:14555] CHIP: [DL] _Init
[1641243251851] [2096:14555] CHIP: [IN] UDP::Init bind&listen port=5542
[1641243251851] [2096:14555] CHIP: [IN] UDP::Init bound to port=5542
[1641243251851] [2096:14555] CHIP: [IN] UDP::Init bind&listen port=5542
[1641243251851] [2096:14555] CHIP: [IN] UDP::Init bound to port=5542
[1641243251851] [2096:14555] CHIP: [IN] TransportMgr initialized
[1641243251851] [2096:14555] CHIP: [DIS] Init fabric pairing table with server storage
[1641243251852] [2096:14555] CHIP: [ZCL] Using ZAP configuration...
[1641243251852] [2096:14555] CHIP: [CTL] System State Initialized...
[1641243251855] [2096:14555] CHIP: [CTL] Generating NOC
[1641243251861] [2096:14555] CHIP: [CTL] Generating ICAC
[1641243251861] [2096:14555] CHIP: [CTL] Generating RCAC
[1641243251862] [2096:14555] CHIP: [DIS] Verifying the received credentials
[1641243251863] [2096:14555] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1641243251863] [2096:14555] CHIP: [IN] 0x04, 0x54, 0x4f, 0x35, 0x09, 0x92, 0xc5, 0x02, 
[1641243251863] [2096:14555] CHIP: [IN] 0xcd, 0xa5, 0x61, 0x12, 0xc8, 0x45, 0x91, 0x06, 
[1641243251863] [2096:14555] CHIP: [IN] 0xcc, 0x06, 0x7a, 0xcf, 0xd7, 0xf9, 0xea, 0x9d, 
[1641243251863] [2096:14555] CHIP: [IN] 0x6e, 0xd3, 0x85, 0x7a, 0x77, 0xae, 0xd6, 0x66, 
[1641243251863] [2096:14555] CHIP: [IN] 0x88, 0xf5, 0x30, 0xc3, 0x35, 0x0b, 0x86, 0xd0, 
[1641243251863] [2096:14555] CHIP: [IN] 0xaf, 0xb8, 0x94, 0xcf, 0x43, 0xa8, 0xe6, 0x28, 
[1641243251863] [2096:14555] CHIP: [IN] 0x32, 0x90, 0x9f, 0xe4, 0x3a, 0x41, 0xae, 0x8b, 
[1641243251863] [2096:14555] CHIP: [IN] 0xf6, 0x57, 0xb4, 0x99, 0xc4, 0x26, 0x52, 0x2c, 
[1641243251863] [2096:14555] CHIP: [IN] 0xe3, 
[1641243251864] [2096:14555] CHIP: [IN] Generated compressed fabric ID
[1641243251864] [2096:14555] CHIP: [IN] 0x63, 0x9e, 0x07, 0x8f, 0x3c, 0x41, 0x10, 0x1d, 
[1641243251864] [2096:14555] CHIP: [DIS] Added new fabric at index: 1, Initialized: 1
[1641243251864] [2096:14555] CHIP: [DIS] Assigned compressed fabric ID: 0x639E078F3C41101D, node ID: 0x000000000001B669
[1641243251864] [2096:14555] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0x639E078F3C41101D
[1641243251864] [2096:14555] CHIP: [IN] UDP::Init bind&listen port=5550
[1641243251864] [2096:14555] CHIP: [IN] UDP::Init bound to port=5550
[1641243251864] [2096:14555] CHIP: [IN] UDP::Init bind&listen port=5550
[1641243251864] [2096:14555] CHIP: [IN] UDP::Init bound to port=5550
[1641243251864] [2096:14555] CHIP: [IN] TransportMgr initialized
[1641243251865] [2096:14555] CHIP: [CTL] Generating NOC
[1641243251865] [2096:14555] CHIP: [CTL] Generating ICAC
[1641243251865] [2096:14555] CHIP: [CTL] Generating RCAC
[1641243251866] [2096:14555] CHIP: [DIS] Verifying the received credentials
[1641243251866] [2096:14555] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1641243251866] [2096:14555] CHIP: [IN] 0x04, 0x54, 0x4f, 0x35, 0x09, 0x92, 0xc5, 0x02, 
[1641243251866] [2096:14555] CHIP: [IN] 0xcd, 0xa5, 0x61, 0x12, 0xc8, 0x45, 0x91, 0x06, 
[1641243251866] [2096:14555] CHIP: [IN] 0xcc, 0x06, 0x7a, 0xcf, 0xd7, 0xf9, 0xea, 0x9d, 
[1641243251866] [2096:14555] CHIP: [IN] 0x6e, 0xd3, 0x85, 0x7a, 0x77, 0xae, 0xd6, 0x66, 
[1641243251866] [2096:14555] CHIP: [IN] 0x88, 0xf5, 0x30, 0xc3, 0x35, 0x0b, 0x86, 0xd0, 
[1641243251866] [2096:14555] CHIP: [IN] 0xaf, 0xb8, 0x94, 0xcf, 0x43, 0xa8, 0xe6, 0x28, 
[1641243251866] [2096:14555] CHIP: [IN] 0x32, 0x90, 0x9f, 0xe4, 0x3a, 0x41, 0xae, 0x8b, 
[1641243251866] [2096:14555] CHIP: [IN] 0xf6, 0x57, 0xb4, 0x99, 0xc4, 0x26, 0x52, 0x2c, 
[1641243251866] [2096:14555] CHIP: [IN] 0xe3, 
[1641243251866] [2096:14555] CHIP: [IN] Generated compressed fabric ID
[1641243251866] [2096:14555] CHIP: [IN] 0xef, 0x49, 0x93, 0x50, 0xa2, 0x2f, 0xda, 0x99, 
[1641243251867] [2096:14555] CHIP: [DIS] Added new fabric at index: 2, Initialized: 1
[1641243251867] [2096:14555] CHIP: [DIS] Assigned compressed fabric ID: 0xEF499350A22FDA99, node ID: 0x000000000001B669
[1641243251867] [2096:14555] CHIP: [CTL] Joined the fabric at index 2. Compressed fabric ID is: 0xEF499350A22FDA99
[1641243251867] [2096:14555] CHIP: [IN] UDP::Init bind&listen port=5550
[1641243251867] [2096:14555] CHIP: [IN] UDP::Init bound to port=5550
[1641243251867] [2096:14555] CHIP: [IN] UDP::Init bind&listen port=5550
[1641243251867] [2096:14555] CHIP: [IN] UDP::Init bound to port=5550
[1641243251867] [2096:14555] CHIP: [IN] TransportMgr initialized
[1641243251868] [2096:14555] CHIP: [CTL] Generating NOC
[1641243251868] [2096:14555] CHIP: [CTL] Generating ICAC
[1641243251868] [2096:14555] CHIP: [CTL] Generating RCAC
[1641243251868] [2096:14555] CHIP: [DIS] Verifying the received credentials
[1641243251869] [2096:14555] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000003 and root pubkey
[1641243251869] [2096:14555] CHIP: [IN] 0x04, 0x54, 0x4f, 0x35, 0x09, 0x92, 0xc5, 0x02, 
[... 92 lines cut ...]
[1641243355820] [2600:18531] CHIP: [IN] 0x04, 0x54, 0x4f, 0x35, 0x09, 0x92, 0xc5, 0x02, 
[1641243355820] [2600:18531] CHIP: [IN] 0xcd, 0xa5, 0x61, 0x12, 0xc8, 0x45, 0x91, 0x06, 
[1641243355820] [2600:18531] CHIP: [IN] 0xcc, 0x06, 0x7a, 0xcf, 0xd7, 0xf9, 0xea, 0x9d, 
[1641243355820] [2600:18531] CHIP: [IN] 0x6e, 0xd3, 0x85, 0x7a, 0x77, 0xae, 0xd6, 0x66, 
[1641243355820] [2600:18531] CHIP: [IN] 0x88, 0xf5, 0x30, 0xc3, 0x35, 0x0b, 0x86, 0xd0, 
[1641243355820] [2600:18531] CHIP: [IN] 0xaf, 0xb8, 0x94, 0xcf, 0x43, 0xa8, 0xe6, 0x28, 
[1641243355820] [2600:18531] CHIP: [IN] 0x32, 0x90, 0x9f, 0xe4, 0x3a, 0x41, 0xae, 0x8b, 
[1641243355820] [2600:18531] CHIP: [IN] 0xf6, 0x57, 0xb4, 0x99, 0xc4, 0x26, 0x52, 0x2c, 
[1641243355820] [2600:18531] CHIP: [IN] 0xe3, 
[1641243355820] [2600:18531] CHIP: [IN] Generated compressed fabric ID
[1641243355820] [2600:18531] CHIP: [IN] 0x45, 0x9a, 0x09, 0x2d, 0xa2, 0x1f, 0x0d, 0xd4, 
[1641243355820] [2600:18531] CHIP: [DIS] Added new fabric at index: 3, Initialized: 1
[1641243355820] [2600:18531] CHIP: [DIS] Assigned compressed fabric ID: 0x459A092DA21F0DD4, node ID: 0x000000000001B669
[1641243355820] [2600:18531] CHIP: [CTL] Joined the fabric at index 3. Compressed fabric ID is: 0x459A092DA21F0DD4
[1641243355820] [2600:18531] CHIP: [IN] UDP::Init bind&listen port=5550
[1641243355820] [2600:18531] CHIP: [IN] UDP::Init bound to port=5550
[1641243355820] [2600:18531] CHIP: [IN] UDP::Init bind&listen port=5550
[1641243355820] [2600:18531] CHIP: [IN] UDP::Init bound to port=5550
[1641243355820] [2600:18531] CHIP: [IN] TransportMgr initialized
[1641243355820] [2600:18541] CHIP: [DL] _OnPlatformEvent
[1641243355820] [2600:18541] CHIP: [TOO] Sending command to node 0x2f59
[1641243365825] [2600:18531] CHIP: [-] ../examples/chip-tool/commands/common/CHIPCommand.cpp:232: CHIP Error 0x00000032: Timeout at ../examples/chip-tool/commands/common/CHIPCommand.cpp:59
[1641243365825] [2600:18531] CHIP: [TOO] Run command failure: ../examples/chip-tool/commands/common/CHIPCommand.cpp:232: CHIP Error 0x00000032: Timeout
[1641243365828] [2600:18531] CHIP: [SPT] VerifyOrDie failure at ../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
'''

tewarid, 2022-01-17:
Title says `15.1` but should be `12.1`. BLE connection works with macOS Monterey 12.2 Beta (21D5039d).

nicolabortignon, 2022-01-17:
Title updated, and yes, I can confirm that on 12.2 Beta 21D5039d, it does fail in the same way you are describing.

tewarid, 2022-01-29:
chip-tool on macOS 12.3 Beta (21E5196i) now able to establish BLE connection. It still fails, but seems to be a different problem, and I also see it on Ubuntu 21.04. I will open a separate issue if I can't find one.

tlykkeberg-grundfos, 2022-02-07:
> Title says `15.1` but should be `12.1`. BLE connection works with macOS Monterey 12.2 Beta (21D5039d) on Intel MacBook Pro (16-inch, 2019), but fails further ahead with
> 
> ```
> [1642419388456] [48885:2429823] CHIP: [BLE] CBManagerState: ON
> [1642419388865] [48885:2429823] CHIP: [BLE] Connecting to device with discriminator: 3840
> [1642419403221] [48885:2429826] CHIP: [BLE] Service not found on the device.
> [1642419403222] [48885:2429826] CHIP: [IN] Clearing BLE pending packets.
> [1642419403222] [48885:2429826] CHIP: [IN] BleConnection Error: ../../third_party/connectedhomeip/src/platform/Darwin/BleConnectionDelegateImpl.mm:221: CHIP Error 0x00000003: Incorrect state
> [1642419418432] [48885:2429826] CHIP: [SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
> [1642419418432] [48885:2429826] CHIP: [TOO] Secure Pairing Failed
> ```

I have the same issue but not on Mac OS (Intel Monterey 12.0.1) but on my iPhone (iOS 15.2.1). We have narrowed it down to an issue with the service not being discovered by the CoreBluetooth Framework. It works perfectly with the DUT on the MacBook, but it fails on the same codebase but on iOS.

danh-geo, 2022-03-01:
I updated to Monterey 12.2.1 today and seeing the same issue. It was working yesterday on Big Sur.

Log attached of attempting to pair with standalone CHIP tool:
[chip-tool-log.txt](https://github.com/project-chip/connectedhomeip/files/8161231/chip-tool-log.txt)

tlykkeberg-grundfos, 2022-03-02:
> 

This is most likely due to iOS (>=15.1) and macOS (>=12.2.1) somehow "swallowing" the Bluetooth Service UUID `0xFFF6` and not letting it onto the application layer. I have tried to change the UUID in both the DUT software and the CHIP stack to something else (i.e. 0x0FF6), which makes it work on iOS and macOS.

ReneJosefsen, 2022-04-05:
This seems to have been fixed in macOS 12.3 (and 12.3.1) so should we close this issue?

@nicolabortignon

bzbarsky-apple, 2022-04-05:
This is resolved in macOS 12.3.

closed 2022-04-05 by bzbarsky-apple

meatherly, 2022-07-15:
I'm using macOS 12.4 and Matter commit b4b340017. For some reason I can commission a device (esp32) using wifi over BLE. But when trying to commission a Thread device (nordic nrf52840dk) with a Thread border router (esp32 ot_br + silabs RCP) over BLE the chip-tool on macOS 12.4 it doesn't work. The chip-tool connects to the thread device over BLE but thread device instantly disconnects with a BLE GAP error. For some reason whenever I use the chip-tool on Ubuntu 20.04 LTS the commissioning process works as expected. I don't have my thread devices with me right now to post the logs but whenever I get back to the office Monday I'll post logs from chip-tool and the thread device. Once I get back I'll post logs from both, the chip-tool on mac  logs and thread device logs.