project-chip/connectedhomeip #13326: Unable to connect via TCP

msandstedt, 2022-01-06:
#### Problem

Code in the sdk ignores the TCP support flag TXT record and always connects by UDP. 
 For instance:

https://github.com/project-chip/connectedhomeip/blob/a49c250c85b334f3f82239f4ca1d5d113373c0e8/src/controller/SetUpCodePairer.cpp#L190

https://github.com/project-chip/connectedhomeip/blob/a49c250c85b334f3f82239f4ca1d5d113373c0e8/src/controller/CHIPDeviceController.cpp#L563

I've tried to modify this and attempted to connect via TCP when this can be supported by both nodes, but this doesn't work.

Specifically, my modifications were:

```
diff --git a/src/controller/CHIPDeviceController.cpp b/src/controller/CHIPDeviceController.cpp
index 18642de99..9e3669830 100644
--- a/src/controller/CHIPDeviceController.cpp
+++ b/src/controller/CHIPDeviceController.cpp
@@ -566,8 +566,16 @@ Transport::PeerAddress DeviceController::ToPeerAddress(const chip::Dnssd::Resolv
     {
         interfaceId = nodeData.mInterfaceId;
     }
-
-    return Transport::PeerAddress::UDP(nodeData.mAddress[0], nodeData.mPort, interfaceId);
+    Transport::PeerAddress peerAddress;
+    if (nodeData.mSupportsTcp)
+    {
+        Transport::PeerAddress::TCP(nodeData.mAddress[0], nodeData.mPort, interfaceId);
+    }
+    else
+    {
+        Transport::PeerAddress::UDP(nodeData.mAddress[0], nodeData.mPort, interfaceId);
+    }
+    return peerAddress;
 }
 
 void DeviceController::OnNodeIdResolved(const chip::Dnssd::ResolvedNodeData & nodeData)
diff --git a/src/controller/SetUpCodePairer.cpp b/src/controller/SetUpCodePairer.cpp
index a53c4fdcb..3c32a0292 100644
--- a/src/controller/SetUpCodePairer.cpp
+++ b/src/controller/SetUpCodePairer.cpp
@@ -187,7 +187,15 @@ void SetUpCodePairer::NotifyCommissionableDeviceDiscovered(const Dnssd::Discover
     LogErrorOnFailure(StopConnectOverSoftAP());
 
     Inet::InterfaceId interfaceId = nodeData.ipAddress[0].IsIPv6LinkLocal() ? nodeData.interfaceId[0] : Inet::InterfaceId::Null();
-    Transport::PeerAddress peerAddress = Transport::PeerAddress::UDP(nodeData.ipAddress[0], nodeData.port, interfaceId);
+    Transport::PeerAddress peerAddress;
+    if (nodeData.supportsTcp)
+    {
+        peerAddress = Transport::PeerAddress::TCP(nodeData.ipAddress[0], nodeData.port, interfaceId);
+    }
+    else
+    {
+        peerAddress = Transport::PeerAddress::UDP(nodeData.ipAddress[0], nodeData.port, interfaceId);
+    }
     RendezvousParameters params        = RendezvousParameters().SetPeerAddress(peerAddress);
     OnDeviceDiscovered(params);
 }
```

This was at this rev:

```
commit d4d57f607f5cc67f349e8aa97a4443e5743595f7
Author: Michael Sandstedt <michael.sandstedt@smartthings.com>
Date:   Tue Jan 4 10:59:55 2022 -0600

    Fix ResolverProxy use-after-free in HandleNodeBrowse (#13291)
```

and with these node types:

commissioner: chip-tool on Darwin
commissionee: all-clusters-app/linux

 #### Proposed Solution

Add missing TCP support to relevant platforms.

woody-apple, 2022-01-21:
I believe TCP was considered optional in the SDK for v1.0, given UDP works with MRP. Tagging for post V1.0

tcarmelveilleux, 2022-10-18:
This is a duplicate of https://github.com/project-chip/connectedhomeip/issues/23028

closed 2022-10-18 by tcarmelveilleux