project-chip/connectedhomeip #13336: chip-tool unable to commission M5 all clusters app

wahajsyed, 2022-01-06:
#### Problem
- chip-tool on raspberry Pi unable to commission M5 stack running all clusters app.

#### Logs:
ubuntu@ubuntu:~/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool pairing ble-wifi 0x12345678 SSID_Test password65 20202021 3840
[1639785253.731323][123238:123238] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-3zeEhT)
[1639785253.732682][123238:123238] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1639785253.733078][123238:123238] CHIP:DL: writing settings to file (/tmp/chip_config.ini-HmYnVS)
[1639785253.733635][123238:123238] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1639785253.734446][123238:123238] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Pc5xqS)
[1639785253.734718][123238:123238] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1639785253.735873][123238:123238] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-8UPZSS)
[1639785253.736890][123238:123238] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1639785253.736976][123238:123238] CHIP:DL: NVS set: chip-counters/reboot-count = 1 (0x1)
[1639785253.737316][123238:123238] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-6yWL1T)
[1639785253.768903][123238:123238] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1639785253.770061][123238:123238] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1639785253.770601][123238:123238] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-8zSYWQ)
[1639785253.771492][123238:123238] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1639785253.771894][123238:123238] CHIP:DL: NVS set: chip-counters/boot-reason = 0 (0x0)
[1639785253.772324][123238:123238] CHIP:DL: writing settings to file (/tmp/chip_config.ini-oeDpsT)
[1639785253.772986][123238:123238] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1639785253.773269][123238:123238] CHIP:DL: NVS set: chip-config/regulatory-location = 0 (0x0)
[1639785253.773638][123238:123238] CHIP:DL: writing settings to file (/tmp/chip_config.ini-KPpdMU)
[1639785253.786642][123238:123238] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1639785253.786740][123238:123238] CHIP:DL: NVS set: chip-config/location-capability = 0 (0x0)
[1639785253.787674][123238:123238] CHIP:DL: Got Ethernet interface: eth0
[1639785253.788410][123238:123238] CHIP:DL: Found the primary Ethernet interface:eth0
[1639785253.789109][123238:123238] CHIP:DL: Got WiFi interface: wlan0
[1639785253.789191][123238:123238] CHIP:DL: Failed to reset WiFi statistic counts
[1639785253.789333][123238:123238] CHIP:IN: UDP::Init bind&listen port=5542
[1639785253.789452][123238:123238] CHIP:IN: UDP::Init bound to port=5542
[1639785253.789508][123238:123238] CHIP:IN: UDP::Init bind&listen port=5542
[1639785253.789614][123238:123238] CHIP:IN: UDP::Init bound to port=5542
[1639785253.789662][123238:123238] CHIP:IN: TransportMgr initialized
[1639785253.789764][123238:123238] CHIP:DIS: Init fabric pairing table with server storage
[1639785253.790219][123238:123238] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Xz0TAU)
[1639785253.791047][123238:123238] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1639785253.791563][123238:123238] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 1000 (0x3E8)
[1639785253.791770][123238:123238] CHIP:ZCL: Using ZAP configuration...
[1639785253.804303][123238:123238] CHIP:DL: Avahi connecting
[1639785253.804427][123238:123238] CHIP:CTL: System State Initialized...
[1639785253.810053][123238:123238] CHIP:CTL: Generating NOC
[1639785253.810488][123238:123238] CHIP:CTL: Generating ICAC
[1639785253.810756][123238:123238] CHIP:CTL: Generating RCAC
[1639785253.811904][123238:123238] CHIP:-: ../../third_party/connectedhomeip/src/platform/Linux/DnssdImpl.cpp:329: CHIP Error 0x00000003: Incorrect state at ../../commands/common/CHIPCommand.cpp:205
[1639785253.812016][123238:123238] CHIP:-: ../../third_party/connectedhomeip/src/platform/Linux/DnssdImpl.cpp:329: CHIP Error 0x00000003: Incorrect state at ../../commands/common/CHIPCommand.cpp:58
[1639785253.812065][123238:123238] CHIP:TOO: Run command failure: ../../third_party/connectedhomeip/src/platform/Linux/DnssdImpl.cpp:329: CHIP Error 0x00000003: Incorrect state
[1639785253.815557][123238:123238] CHIP:SPT: VerifyOrDie failure at ../../third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
Aborted (core dumped)

 #### Proposed Solution
- Raspi ubuntu needs avahi-utils package to function correctly

andy31415, 2022-01-06:
Is this an SDK issue? sounds like it is an issue in preparing a raspberry pi ubuntu image.

bzbarsky-apple, 2022-01-06:
Made it clearer what the issue is about: we should be usefully reporting the error, not just segfaulting.

closed 2023-02-04 by stale[bot]