project-chip/connectedhomeip #13345: [TE-7-Test-Harness] cli.sh available test cases are failing with Google Thermostat

ramakrishnanandyala, 2022-01-06:
Trying to use test-harness-7 tool to execute TE-7 test cases related to thermostat.

**Steps followed :
1. setup test-harness-7 on raspi
2. Commissioned thermostat using test-harness-7's chip-tool  [./chip-tool pairing onnetwork <node_id> <pin code>]
3. Edited wifi_config and dut_config in test-harness.config file as below.
ubuntu@matter-te-7:~$ cat test-harness.config** 
{
  "logging": {
    "path": "./logs",
    "filename": "{time:YYYY-MM-DD}.log",
    "level": "debug",
    "rotation": "20 days",
    "retention": "1 months",
    "format": "<level>{level: <8}</level> | <green>{time:YYYY-MM-DD HH:mm:ss.SSS}</green> | <cyan>{name}</cyan>:<cyan>{function}</cyan>:<cyan>{line}</cyan> | <level>{message}</level>"
  },
  "notifier":{
    "enable_notifications": false,
    "notifier_config":{
      "username": "xiole.chip.test@gmail.com",
      "password": "yntwdxwyofvehafg",
      "to": "xiole.chip.test@gmail.com",
      "subject": "CHIP Tool Crash Log"
    }
  },
  "network":{
    "fabric_id":"0",
    "thread": {
      "dataset": {
        "channel": "15",
        "panid": "0x1234",
        "extpanid": "1111111122222222",
        "masterkey": "00112233445566778899aabbccddeeff",
        "networkname": "DEMO"
      },
      "rcp_serial_path": "/dev/ttyACM0",
      "rcp_baudrate": 115200,
      "on_mesh_prefix": "fd11:22::/64",
      "network_interface": "wlan0"
    },
    "wifi":{
      "ssid": "rkreddy",
      "password": "adhyareddy"
    }
  },
  "dut_config":{
    "pairing_mode":"onnetwork",
    "setup_code":"12958229",
    "discriminator":"3840",
    "ip_address":"10.0.0.245",
    "port":"5540"
  }
}

**4. from $cli, executed TC_DM_1.1 test case and i see below error
command ---> [./cli.sh run-tests --selected-tests '{"chip_tool_tests":{"FirstChipToolSuite":{"TC_DM_1_1":1}}}']**

**Error**
INFO     | 2021-12-29 22:23:30.766 | Run Test Runner is Ready
INFO     | 2021-12-29 22:23:30.771 | Test Run Executing
INFO     | 2021-12-29 22:23:30.783 | Test Suite Executing: Test Suite run with chip-tool tests
INFO     | 2021-12-29 22:23:30.785 | Setting up chip_tool
INFO     | 2021-12-29 22:25:33.216 | Pair with DUT
CHIPTOOL | 2021-12-29 22:25:33.218 | [1640816611.365406][8:8] CHIP:CTL: Generating NOC
CHIPTOOL | 2021-12-29 22:25:33.219 | [1640816611.365756][8:8] CHIP:CTL: Generating ICAC
CHIPTOOL | 2021-12-29 22:25:33.220 | [1640816611.366031][8:8] CHIP:CTL: Generating RCAC
CHIPTOOL | 2021-12-29 22:25:33.222 | [1640816611.375033][8:8] CHIP:DL: AUDIT: ===== RANDOM NUMBER GENERATOR AUDIT START ====
CHIPTOOL | 2021-12-29 22:25:33.223 | [1640816611.375077][8:8] CHIP:DL: AUDIT: * Validate buf1 and buf2 are <<<different every run/boot!>>>
CHIPTOOL | 2021-12-29 22:25:33.224 | [1640816611.375094][8:8] CHIP:DL: AUDIT: * Validate r1 and r2 are <<<different every run/boot!>>>
CHIPTOOL | 2021-12-29 22:25:33.226 | [1640816611.375111][8:8] CHIP:DL: AUDIT: * buf1: EC1BFAE170938194B9DD0C5EFE1523A1
[... 57 lines cut ...]
INFO     | 2021-12-29 22:25:33.350 | Cancel test step
INFO     | 2021-12-29 22:25:33.352 | Cancel test step
INFO     | 2021-12-29 22:25:33.355 | Cancel test step
INFO     | 2021-12-29 22:25:33.357 | Cancel test step
INFO     | 2021-12-29 22:25:33.359 | Cancel test step
INFO     | 2021-12-29 22:25:33.362 | Cancel test step
INFO     | 2021-12-29 22:25:33.364 | Cancel test step
INFO     | 2021-12-29 22:25:33.366 | Cancel test step
INFO     | 2021-12-29 22:25:33.369 | Cancel test step
INFO     | 2021-12-29 22:25:33.371 | Cancel test step
INFO     | 2021-12-29 22:25:33.373 | Cancel test step
INFO     | 2021-12-29 22:25:33.375 | Cancel test step
INFO     | 2021-12-29 22:25:33.377 | Cancel test step
INFO     | 2021-12-29 22:25:33.380 | Cancel test step
INFO     | 2021-12-29 22:25:33.385 | Cancel test step
INFO     | 2021-12-29 22:25:33.388 | Cancel test step
INFO     | 2021-12-29 22:25:33.390 | Cancel test step
INFO     | 2021-12-29 22:25:33.392 | Cancel test step
INFO     | 2021-12-29 22:25:33.395 | Stopping chip-tool container
INFO     | 2021-12-29 22:25:33.397 | Test Suite Completed [ERROR]: Test Suite run with chip-tool tests
INFO     | 2021-12-29 22:25:33.405 | Test Run Completed [ERROR]

Issue :
In test-harness.config file under dut_config I don't see these parameters (node_id and pin code) but I see TE-6 supported parameters (setup code, discriminatior, ip address, port).
Is test-harness should be updated or need any extra parameters to be added?

mikaelhm, 2022-01-06:
Node-id Is randomly generated.

Can you attach `backend/logs/<date>.log` as well?

ramakrishnanandyala, 2022-01-06:
this is the o/p i got from cli.sh command i didn't get any backend/logs/.. date.log folder created.

ubuntu@matter-te-7:~/cli$ ./cli.sh run-tests --selected-tests '{"chip_tool_tests":{"FirstChipToolSuite":{"TC_DM_1_1":1}}}'
Creating new test run with title: 2021-12-29 22:23:29.319018
Starting Test run: Title: 2021-12-29 22:23:29.319018, id: 72
Test Run [EXECUTING]
  - Test Suite run with chip-tool tests [EXECUTING]
  - Test Suite run with chip-tool tests [EXECUTING]
      - 10.1.1. [TC-DM-1.1] Basic Cluster Server Attributes [CANCELLED]
            - Starting ChipTool Test [CANCELLED]
            - Query Interaction Model Version [CANCELLED]
            - Query Vendor Name [CANCELLED]
            - Query VendorID [CANCELLED]
            - Query Product Name [CANCELLED]
            - Query ProductID [CANCELLED]
            - Query User Label [CANCELLED]
            - Query User Location [CANCELLED]
            - Query HardwareVersion [CANCELLED]
            - Query HardwareVersionString [CANCELLED]
            - Query SoftwareVersion [CANCELLED]
            - Query SoftwareVersionString [CANCELLED]
            - Query ManufacturingDate [CANCELLED]
            - Query PartNumber [CANCELLED]
            - Query ProductURL [CANCELLED]
            - Query ProductLabel [CANCELLED]
            - Query SerialNumber [CANCELLED]
            - Query LocalConfigDisabled [CANCELLED]
            - Query Reachable [CANCELLED]
  - Test Suite run with chip-tool tests [ERROR]
Test Run [ERROR]
Log output in: './output_logs/test_run_2021-12-29 22:23:29.319018.log'

mikaelhm, 2022-01-06:
its in `~/chip-certification-tool/backend/logs/2021-12-29.log`

ramakrishnanandyala, 2022-01-06:
got it

ubuntu@matter-te-7:~/chip-certification-tool/backend/logs$ cat 2021-12-29.log 
INFO     | 2021-12-29 03:22:44.450 | uvicorn.server:shutdown:263 | Shutting down
INFO     | 2021-12-29 03:22:44.554 | uvicorn.lifespan.on:shutdown:64 | Waiting for application shutdown.
INFO     | 2021-12-29 03:22:44.557 | uvicorn.lifespan.on:shutdown:75 | Application shutdown complete.
INFO     | 2021-12-29 03:22:44.561 | uvicorn.server:serve:94 | Finished server process [15]
INFO     | 2021-12-29 03:24:02.220 | app.log:__configure_notifier_handler:89 | Exception notifier not enabled
INFO     | 2021-12-29 03:24:02.224 | uvicorn.server:serve:84 | Started server process [15]
INFO     | 2021-12-29 03:24:02.228 | uvicorn.lifespan.on:startup:45 | Waiting for application startup.
INFO     | 2021-12-29 03:24:02.232 | uvicorn.lifespan.on:startup:59 | Application startup complete.
INFO     | 2021-12-29 03:32:55.690 | app.log:__configure_notifier_handler:89 | Exception notifier not enabled
INFO     | 2021-12-29 03:32:55.693 | uvicorn.server:serve:84 | Started server process [16]
INFO     | 2021-12-29 03:32:55.695 | uvicorn.lifespan.on:startup:45 | Waiting for application startup.
INFO     | 2021-12-29 03:32:55.697 | uvicorn.lifespan.on:startup:59 | Application startup complete.
INFO     | 2021-12-29 03:39:46.467 | uvicorn.protocols.http.httptools_impl:send:440 | 172.18.0.4:54548 - "GET /api/v1/test_collections/ HTTP/1.1" 200
INFO     | 2021-12-29 03:40:33.423 | uvicorn.protocols.http.httptools_impl:send:440 | 172.18.0.4:54554 - "GET /api/v1/test_collections/ HTTP/1.1" 200
INFO     | 2021-12-29 21:52:13.291 | uvicorn.protocols.http.httptools_impl:send:440 | 172.18.0.4:54578 - "POST /api/v1/test_run_executions/ HTTP/1.1" 200
INFO     | 2021-12-29 21:52:13.330 | uvicorn.protocols.websockets.websockets_impl:asgi_send:231 | ('172.18.0.4', 54586) - "WebSocket /api/v1/ws" [accepted]
DEBUG    | 2021-12-29 21:52:13.333 | app.test_engine.test_runner:state:122 | Setting TestRunner state to LOADING
INFO     | 2021-12-29 21:52:13.336 | websockets.legacy.server:handshake:633 | connection open
INFO     | 2021-12-29 21:52:13.359 | app.test_engine.test_runner:load_test_run:145 | Loading Test Run: 2021-12-29 21:52:12.856383
INFO     | 2021-12-29 21:52:13.360 | app.test_engine.test_runner:load_test_run:148 | Loading Test Run: 2021-12-29 21:52:12.856383
DEBUG    | 2021-12-29 21:52:13.391 | app.test_engine.test_runner:state:122 | Setting TestRunner state to READY
INFO     | 2021-12-29 21:52:13.393 | app.test_engine.test_runner:load_test_run:157 | Test Runner is Ready
INFO     | 2021-12-29 21:52:13.423 | uvicorn.protocols.http.httptools_impl:send:440 | 172.18.0.4:54578 - "POST /api/v1/test_run_executions/67/start HTTP/1.1" 200
INFO     | 2021-12-29 21:52:13.425 | app.test_engine.test_runner:run:195 | Run Test Runner is Ready
DEBUG    | 2021-12-29 21:52:13.427 | app.test_engine.test_runner:state:122 | Setting TestRunner state to RUNNING
DEBUG    | 2021-12-29 21:52:13.429 | app.test_engine.test_db_observer:dispatch:26 | Received Dispatch event
DEBUG    | 2021-12-29 21:52:13.431 | app.test_engine.test_db_observer:__onTestRunUpdate:38 | Test Run Observer received
DEBUG    | 2021-12-29 21:52:13.448 | app.test_engine.test_ui_observer:dispatch:27 | Received Dispatch event
DEBUG    | 2021-12-29 21:52:13.450 | app.test_engine.test_ui_observer:__onTestRunUpdate:39 | Test Run Observer received
INFO     | 2021-12-29 21:52:13.453 | app.test_engine.models.test_run:mark_as_executing:71 | Test Run Executing
DEBUG    | 2021-12-29 21:52:13.458 | app.test_engine.test_db_observer:dispatch:26 | Received Dispatch event
DEBUG    | 2021-12-29 21:52:13.459 | app.test_engine.test_db_observer:__onTestSuiteUpdate:49 | Test Suite Observer received
DEBUG    | 2021-12-29 21:52:13.476 | app.test_engine.test_ui_observer:dispatch:27 | Received Dispatch event
DEBUG    | 2021-12-29 21:52:13.478 | app.test_engine.test_ui_observer:__onTestSuiteUpdate:50 | Test Suite Observer received
INFO     | 2021-12-29 21:52:13.480 | app.test_engine.models.test_suite:mark_as_executing:105 | Test Suite Executing: Test Suite run with chip-tool tests
INFO     | 2021-12-29 21:52:13.482 | app.chip_tool.test_suite:setup:28 | Setting up chip_tool
DEBUG    | 2021-12-29 21:52:13.794 | urllib3.connectionpool:_make_request:452 | http://localhost:None "POST /v1.41/containers/create HTTP/1.1" 201 88
DEBUG    | 2021-12-29 21:52:13.804 | urllib3.connectionpool:_make_request:452 | http://localhost:None "GET /v1.41/containers/ac8fa60ae0fe8da94c9a9113118fea9a048b3d6438653dcd225aea0435cd9aec/json HTTP/1.1" 200 None
DEBUG    | 2021-12-29 21:52:14.671 | urllib3.connectionpool:_make_request:452 | http://localhost:None "POST /v1.41/containers/ac8fa60ae0fe8da94c9a9113118fea9a048b3d6438653dcd225aea0435cd9aec/start HTTP/1.1" 204 0
DEBUG    | 2021-12-29 21:52:14.689 | urllib3.connectionpool:_make_request:452 | http://localhost:None "GET /v1.41/containers/ac8fa60ae0fe8da94c9a9113118fea9a048b3d6438653dcd225aea0435cd9aec/json HTTP/1.1" 200 None
INFO     | 2021-12-29 21:52:14.693 | app.generic_controller.container_manager:create_container:28 | Container running for chip-tool:latest
INFO     | 2021-12-29 21:52:14.696 | app.chip_tool.chip_tool:start_container:93 | New Random node-id generated: 0x2dd76679
INFO     | 2021-12-29 21:52:14.699 | app.chip_tool.chip_tool:start_container:95 | 
            chip-tool started: ac8fa60ae0fe
[... cut ...]
DEBUG    | 2021-12-29 22:53:49.610 | app.test_engine.test_db_observer:dispatch:26 | Received Dispatch event
DEBUG    | 2021-12-29 22:53:49.611 | app.test_engine.test_db_observer:__onTestRunUpdate:38 | Test Run Observer received
INFO     | 2021-12-29 22:53:49.624 | app.test_engine.models.test_run:mark_as_completed:67 | Test Run Completed [ERROR]
DEBUG    | 2021-12-29 22:53:49.626 | app.test_engine.test_runner:state:122 | Setting TestRunner state to IDLE
INFO     | 2021-12-29 22:53:49.798 | websockets.legacy.server:handler:252 | connection closed

ramakrishnanandyala, 2022-01-06:
Created same issue with right path here 
https://github.com/CHIP-Specifications/chip-certification-tool/issues/268

andy31415, 2022-05-30:
@ramakrishnanandyala - could you check ToT for this? TE7 is very old code.

Also could you re-state the description of the test cases in a shorter format? This PR has a lot of pasted logs, it is hard to look through them to isolate what is needed here. see https://github.com/project-chip/connectedhomeip/blob/master/docs/BUG_REPORT.md

closed 2022-12-20 by stale[bot]