project-chip/connectedhomeip #13362: k32w STL compile error due to that it defines `global`

kghost, 2022-01-07:
#### Problem
https://github.com/project-chip/connectedhomeip/runs/4735883752?check_suite_focus=true

In `./middleware/wireless/bluetooth/host/interface/ble_utils.h` inside k32w sdk line 77-79
```
/*! Marks that this variable is in the interface. */    
#ifndef global                                                                                                                                                                                                     
#define global    
#endif    
```

In `.environment/cipd/pigweed/arm-none-eabi/include/c++/10.2.1/bits/locale_classes.h` line 300 we have
```
    static locale    
    global(const locale& __loc);                                                                                                                                                                                   
```

After macro preprocessor it becomes:

```
# 300 ".environment/cipd/pigweed/arm-none-eabi/include/c++/10.2.1/bits/locale_classes.h" 3
    static locale                                                                                                                                                                                                  
    (const locale& __loc);
```

Triggers a compile error.

 #### Proposed Solution
Fix in k32w sdk

andy31415, 2022-01-07:
@kghost - including locale may be a problem due to sizes. While making k32w not being able to  do stl may be an issue, maybe we can also solve by using a restricted stl subset inside matter in general.

andy31415, 2022-01-07:
Include chain seems to be:

```
2022-01-07 06:27:44 INFO    In file included from /__w/connectedhomeip/connectedhomeip/.environment/cipd/pigweed/arm-none-eabi/include/c++/10.2.1/bits/ios_base.h:41,
2022-01-07 06:27:44 INFO                     from /__w/connectedhomeip/connectedhomeip/.environment/cipd/pigweed/arm-none-eabi/include/c++/10.2.1/streambuf:41,
2022-01-07 06:27:44 INFO                     from /__w/connectedhomeip/connectedhomeip/.environment/cipd/pigweed/arm-none-eabi/include/c++/10.2.1/bits/streambuf_iterator.h:35,
2022-01-07 06:27:44 INFO                     from /__w/connectedhomeip/connectedhomeip/.environment/cipd/pigweed/arm-none-eabi/include/c++/10.2.1/iterator:66,
2022-01-07 06:27:44 INFO                     from ../../examples/lighting-app/nxp/k32w/k32w0/third_party/connectedhomeip/src/lib/support/IntrusiveList.h:19,
2022-01-07 06:27:44 INFO                     from ../../examples/lighting-app/nxp/k32w/k32w0/third_party/connectedhomeip/src/transport/SessionHolder.h:20,
```

It seems like including iterator pulls in ios_base and streambuf, which we try to avoid within  the entire project. I would propose to make IntrusiveList not depend in `<iterator>`

kghost, 2022-01-07:
It takes me a lot of time to find the root cause of the problem, I don't want to waste someone's time for this problem again. these headers are included by lots of STL parts. Defining `global` is obviously not robust, and will cause problems.

Including `streambuf` won't increase binary size as long as it is not linked. There is no reason to avoid it.

andy31415, 2022-01-07:
I agree that k32w is an issue with the NXP SDK that should be fixed.

However for sizes: #13330 shows a flash increase of  100K+ for efr32 and p6 and about 60K for qpg and telink.
I believe this may be in part due to locale. I stand by my claim that we should not include headers that may include anything related to locale as a sidefffect.

This effectively means usage of `<iterator>` should be banned in the matter sdk.

andy31415, 2022-01-07:
After size checks in 13330 it looks like the size increase was due to debug info and not locale as I assumed. So including iterator seems usable and does not need banning.

I am still unsure which path is faster: updating iterator usage in matter sdk or fixing k32w SDK. I suspect matter sdk updates may be faster.

andy31415, 2022-01-07:
third_party/nxp/k32w0_sdk/sdk_fixes/patch_k32w_sdk.sh has some sdk updates which we could use to apply even more patches for chip.

bzbarsky-apple, 2022-01-08:
It looks like #13330 snuck in a fix for this?  Was that purposeful?  It's really odd to fix this inside that PR....  @kghost

woody-apple, 2022-01-21:
Marking as resolved, given the above comment.

closed 2022-01-21 by woody-apple