project-chip/connectedhomeip #13363: [cross-compile][linux-arm] avahi-client  not found

standback, 2022-01-07:
Hi guys.
i try to do cross-compile,  target on a linux-arm   platform.

when i run "ninja -C out/arm". 
a error popup 
```
Package avahi-client was not found in the pkg-config search path.
Perhaps you should add the directory containing `avahi-client.pc'
to the PKG_CONFIG_PATH environment variable
No package 'avahi-client' found
Could not run pkg-config.
```

according to nomal compile .
the building doc  list some requested package to install first, like below
```
sudo apt-get install git gcc g++ python pkg-config libssl-dev libdbus-1-dev \
     libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev \
     python3-pip unzip libgirepository1.0-dev libcairo2-dev
```

it shows libavahi-client-dev package should be installed in the host.

but i am do cross-compile now. PKG-CONFIG_PATH is set to target-toolchain path.
and i think there should be a cross-version  of avahi-client in target-toolchain.
the fact is  target toolchain do not include. 

what should i do,  do i need to use the target toolchain to install avahi-client?

caipiblack, 2022-01-07:
Hi,

To cross compile matter for a specific hardware running on Linux you need some libraries compiled for this system.

For example: `libavahi` and `glib-2.0` 

With OpenWRT it's easy to build these libraries because they are available in OpenWRT packages. This is the same for Yocto.

If the destination system doesn't provide these libraries or something to "auto build" these libraries, then I think you will have to build these libraries by hand. In this case you also need to provide/build the libraries needed to build these libraries.. (For example `libffi`, `libiconv`, `libevent`, `libdaemon`, `libdbus`)

standback, 2022-01-10:
thanks for your reply.

it seems that it is a little bit complicated to do cross compile.

thanls

closed 2022-01-10 by standback