project-chip/connectedhomeip #13389: Chip tool / Python Controller not allowing to read/write to invalid attribute

Garik098, 2022-01-08:
While using Chip Tool / Python Controller to read/write invalid attribute from an accessory, we are seeing error at the Controller (Chip Tool / Python) side. 

Expectation: Controller shouldn't send an error and the request should go through to Accessory. The error should come from the Accessory saying that the attribute is invalid. 

Below is an example of reading testingdata attribute which is invalid. The error shown at the end is coming from Chip Tool.

-----------------------
ubuntu@matter-te-6:~/TE7/connectedhomeip/out/debug$ ./chip-tool basic read testingdata
[1641600084.085781][16564:16564] CHIP:TOO: Unknown attribute: testingdata
Usage:
  ./chip-tool basic read attribute-name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Attributes:                                                                         |
  +-------------------------------------------------------------------------------------+
  | * interaction-model-version                                                         |
  | * vendor-name                                                                       |
  | * vendor-id                                                                         |
  | * product-name                                                                      |
  | * product-id                                                                        |
  | * node-label                                                                        |
  | * location                                                                          |
  | * hardware-version                                                                  |
  | * hardware-version-string                                                           |
  | * software-version                                                                  |
  | * software-version-string                                                           |
  | * manufacturing-date                                                                |
  | * part-number                                                                       |
  | * product-url                                                                       |
  | * product-label                                                                     |
  | * serial-number                                                                     |
  | * local-config-disabled                                                             |
  | * reachable                                                                         |
  | * unique-id                                                                         |
  | * attribute-list                                                                    |
  | * cluster-revision                                                                  |
  +-------------------------------------------------------------------------------------+
[1641600084.086341][16564:16564] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/Commands.cpp:107: Error 0x0000002F
----------------------

bzbarsky-apple, 2022-01-08:
For a _read_, this could in theory work in that we could use a low-level reading API that does not attempt to decode the response data itself and assert that we do not get back data at all.

For a _write_ things are more complicated: to write an attribute you have to know what type it is, so you can encode the data on the wire.  But for an attribute that's not defined anywhere, there's no way to know the type, right?

cjandhyala, 2022-01-12:
@bzbarsky-apple correct, sender can send any type of data. This is for the attribute that is not there in the spec.

bzbarsky-apple, 2022-01-12:
How is the sender expecting the data to get encoded?  That is, how does the sender describe that since the SDK has no idea how to do it for an attribute it has no schema for?

vivien-apple, 2022-01-27:
`chip-tool` does it now with the `write-by-id` command. You need to provide the attribute id and you can build "packet" manually with any type of data (list, struct, list of structs, etc...)

Similarly you can do `command-by-id`, `read-by-id`, `subscribe-by-id`, `read-event-by-id`,  `subscribe-event-by-id`

For example you can do:
```
$ ./out/debug/standalone/chip-tool onoff read-by-id 0 0x12345 1 
$ ./out/debug/standalone/chip-tool onoff write-by-id 0x00004001 '5' 0x12345 1
$ ./out/debug/standalone/chip-tool testcluster command-by-id 0 "{}" 0x12345 1
```

You can even try to combine that with wildcards if you want to have fun...
```
$ ./out/debug/standalone/chip-tool onoff write-by-id 0xffffffff '5' 0x12345 0xffff # all attributes on all endpoints
```

closed 2022-01-27 by vivien-apple