project-chip/connectedhomeip #13418: In wifidiagnostic cluster error occured during execution of test cases using TE7.5 image V 1.5.

somu1710, 2022-01-10:
While executing wifidiagnostic cluster we getting error during execution of some commands using new image TE7.5 of this commit id b0af6ba42b6ce494748f18ecf3e054a400b8f45b
 * expected behavior - success
 * actual behavior - Aborted ( core dumped)
 * actual logs attached here
 Failing commands -./chip-tool wifinetworkdiagnostics read channel-number 1 0
                                 ./chip-tool wifinetworkdiagnostics read beacon-lost-count 1 0
                                  ./chip-tool wifinetworkdiagnostics read rssi 1 0
                                  ./chip-tool wifinetworkdiagnostics read current-max-rate 1 0
[Logs fail cmmds of wifidiag.txt](https://github.com/project-chip/connectedhomeip/files/7840880/fail.cmmds.of.wifidiag.txt)

yufengwangca, 2022-01-14:
It looks you use invalid nodeid  '1' in the command, could you try with the target nodeid, such as:

./chip-tool wifinetworkdiagnostics read channel-number 12344321 0

Survensa, 2022-01-17:
@yufengwangca 
 yeah, I tried with the node id what u mentioned in the comment above (12344321),its gets same result what i get previously by using node id 1.
here i attached the log file below for your reference.
[wifidiag fail logs.txt](https://github.com/project-chip/connectedhomeip/files/7880263/wifidiag.fail.logs.txt)

somu1710, 2022-01-17:
@yufengwangca 
 yeah, I tried with the node id what u mentioned in the comment above (12344321),its gets same result what i get previously by using node id 1.
here i attached the log file below for your reference.
[wifidiag fail logs.txt](https://github.com/project-chip/connectedhomeip/files/7880290/wifidiag.fail.logs.txt)

liamgonyea, 2022-01-21:
### Update for TE 7.5: Same Attribute Failures and Additional Command Failure
commit: `b0af6ba42b6ce494748f18ecf3e054a400b8f45b`

Same issue persists across the entire issue filed above as well as an additional command failing for `TC-WIFIDIAG-3.1`.

TC-WIFIDIAG-1.1 & TC-WIFIDIAG-1.2 issues filed already here: #11652
TC-WIFIDIAG-3.1 issue filed already here: #11653

#### Affected Attributes: 
1. channel-number 
2. rssi
3. beacon-lost-count
4. current-max-rate

#### Affected Command: 
5. ./chip-tool wifinetworkdiagnostics reset-counts 1234 0

#### Test Cases
TC-WIFIDIAG-1.1 | Attributes with server as DUT 
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/wifi_diagnostics.adoc#tc-wifidiag-1-1-attributes-with-server-as-dut

TC-WIFIDIAG-1.2 | Attributes observation at client as DUT 
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/wifi_diagnostics.adoc#tc-wifidiag-1-2-attributes-observation-at-client-as-dut

TC-WIFIDIAG-3.1 | Command received functionality with server as DUT
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/wifi_diagnostics.adoc#tc-wifidiag-3-1-command-received-functionality-with-server-as-dut

### Lines of Interest from Logs
(From `consolidated_mac_accessory_TC-WIFIDIAG-1-1.txt` only, as `consolidated_mac_accessory_TC-WIFIDIAG-1-2.txt` logs are the same commands and same failures)
1. channel-number 
> [1642731832650] [90506:3876295] CHIP: [TOO] Run command failure: ../../commands/common/CHIPCommand.cpp:232: CHIP Error 0x00000032: Timeout
2. rssi
> [1642731877931] [90509:3876831] CHIP: [TOO] Run command failure: ../../commands/common/CHIPCommand.cpp:232: CHIP Error 0x00000032: Timeout
3. beacon-lost-count
> [1642731923340] [90512:3877318] CHIP: [TOO] Run command failure: ../../commands/common/CHIPCommand.cpp:232: CHIP Error 0x00000032: Timeout
4. current-max-rate
> [1642732146056] [90529:3879224] CHIP: [TOO] Run command failure: ../../commands/common/CHIPCommand.cpp:232: CHIP Error 0x00000032: Timeout

(From `consolidated_mac_accessory_TC-WIFIDIAG-3-1.txt`)
5. ./chip-tool wifinetworkdiagnostics reset-counts 1234 0
> [1642732716954] [90578:3884772] CHIP: [TOO] Run command failure: ../../third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2502: CHIP Error 0x000000AC: Internal error

### Logs
[consolidated_mac_accessory_TC-WIFIDIAG-1-1.txt](https://github.com/project-chip/connectedhomeip/files/7916208/consolidated_mac_accessory_TC-WIFIDIAG-1-1.txt)
[consolidated_mac_accessory_TC-WIFIDIAG-1-2.txt](https://github.com/project-chip/connectedhomeip/files/7916209/consolidated_mac_accessory_TC-WIFIDIAG-1-2.txt)
[consolidated_mac_accessory_TC-WIFIDIAG-3-1.txt](https://github.com/project-chip/connectedhomeip/files/7916274/consolidated_mac_accessory_TC-WIFIDIAG-3-1.txt)

yufengwangca, 2022-02-03:
@liamgonyea 

From all logs you posted, I saw the following error message:

```
[1642732716954] [90578:3884772] CHIP: [-] ../../third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2502: CHIP Error 0x000000AC: Internal error at ../../commands/common/CHIPCommand.cpp:59
[1642732716954] [90578:3884772] CHIP: [TOO] Run command failure: ../../third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2502: CHIP Error 0x000000AC: Internal error
[1642732716957] [90578:3884772] CHIP: [SPT] VerifyOrDie failure at ../../third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
```
1. Does the error only occurs on Commissioner side(chip-tool) with program crash?
2. On which platform you are running the device side example app? We only implement those diagnostic cluster on Linux platform

yufengwangca, 2022-02-09:
@somu1710 

Please note, the Wifi diagnostic cluster is only implemented on Linux reference platform on endpoint 0. I am not sure if TE7.5 branch contain the latest bug fixes, but I am not able to see any of issue you mentioned on ToT, could you confirm.

Here is the steps I use to test those attributes:

```
On Server Side:
sudo rm -rf /tmp/chip_*
./chip-all-clusters-app

On Client Side:
sudo rm /tmp/chip_*
./chip-tool pairing onnetwork 12344321 20202021

yufengw@yufengw-SEi:~/connectedhomeip/out/debug/standalone$ ./chip-tool wifinetworkdiagnostics read channel-number 12344321 0
......
[1644386085.373221][143947:143952] CHIP:DMG: ReportDataMessage =
[1644386085.373227][143947:143952] CHIP:DMG: {
[1644386085.373231][143947:143952] CHIP:DMG: 	AttributeReportIBs =
[1644386085.373239][143947:143952] CHIP:DMG: 	[
[1644386085.373243][143947:143952] CHIP:DMG: 		AttributeReportIB =
[1644386085.373251][143947:143952] CHIP:DMG: 		{
[1644386085.373256][143947:143952] CHIP:DMG: 			AttributeDataIB =
[1644386085.373262][143947:143952] CHIP:DMG: 			{
[1644386085.373267][143947:143952] CHIP:DMG: 				DataVersion = 0x82208c89,
[1644386085.373272][143947:143952] CHIP:DMG: 				AttributePathIB =
[1644386085.373278][143947:143952] CHIP:DMG: 				{
[1644386085.373286][143947:143952] CHIP:DMG: 					Endpoint = 0x0,
[1644386085.373298][143947:143952] CHIP:DMG: 					Cluster = 0x36,
[1644386085.373305][143947:143952] CHIP:DMG: 					Attribute = 0x0000_0003,
[1644386085.373311][143947:143952] CHIP:DMG: 				}
[1644386085.373323][143947:143952] CHIP:DMG: 					
[1644386085.373332][143947:143952] CHIP:DMG: 					Data = 149, 
[1644386085.373341][143947:143952] CHIP:DMG: 			},
[1644386085.373349][143947:143952] CHIP:DMG: 			
[1644386085.373354][143947:143952] CHIP:DMG: 		},
[1644386085.373362][143947:143952] CHIP:DMG: 		
[1644386085.373367][143947:143952] CHIP:DMG: 	],
[1644386085.373375][143947:143952] CHIP:DMG: 	
[1644386085.373381][143947:143952] CHIP:DMG: 	SuppressResponse = true, 
[1644386085.373387][143947:143952] CHIP:DMG: 	InteractionModelRevision = 1
[1644386085.373391][143947:143952] CHIP:DMG: }
[1644386085.373441][143947:143952] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0036 Attribute 0x0000_0003
[1644386085.373489][143947:143952] CHIP:TOO:   ChannelNumber: 149

yufengw@yufengw-SEi:~/connectedhomeip/out/debug/standalone$ ./chip-tool wifinetworkdiagnostics read rssi 12344321 0
......
[1644386104.508053][143955:143960] CHIP:DMG: ReportDataMessage =
[1644386104.508057][143955:143960] CHIP:DMG: {
[1644386104.508062][143955:143960] CHIP:DMG: 	AttributeReportIBs =
[1644386104.508069][143955:143960] CHIP:DMG: 	[
[1644386104.508072][143955:143960] CHIP:DMG: 		AttributeReportIB =
[1644386104.508079][143955:143960] CHIP:DMG: 		{
[1644386104.508083][143955:143960] CHIP:DMG: 			AttributeDataIB =
[1644386104.508088][143955:143960] CHIP:DMG: 			{
[1644386104.508094][143955:143960] CHIP:DMG: 				DataVersion = 0x82208c89,
[1644386104.508098][143955:143960] CHIP:DMG: 				AttributePathIB =
[1644386104.508104][143955:143960] CHIP:DMG: 				{
[1644386104.508109][143955:143960] CHIP:DMG: 					Endpoint = 0x0,
[1644386104.508133][143955:143960] CHIP:DMG: 					Cluster = 0x36,
[1644386104.508140][143955:143960] CHIP:DMG: 					Attribute = 0x0000_0004,
[1644386104.508144][143955:143960] CHIP:DMG: 				}
[1644386104.508149][143955:143960] CHIP:DMG: 					
[1644386104.508154][143955:143960] CHIP:DMG: 					Data = -54, 
[1644386104.508159][143955:143960] CHIP:DMG: 			},
[1644386104.508165][143955:143960] CHIP:DMG: 			
[1644386104.508168][143955:143960] CHIP:DMG: 		},
[1644386104.508173][143955:143960] CHIP:DMG: 		
[1644386104.508176][143955:143960] CHIP:DMG: 	],
[... 45 lines cut ...]
[1644386411.432839][144031:144036] CHIP:DMG: 		{
[1644386411.432843][144031:144036] CHIP:DMG: 			AttributeDataIB =
[1644386411.432849][144031:144036] CHIP:DMG: 			{
[1644386411.432855][144031:144036] CHIP:DMG: 				DataVersion = 0x82208c89,
[1644386411.432859][144031:144036] CHIP:DMG: 				AttributePathIB =
[1644386411.432864][144031:144036] CHIP:DMG: 				{
[1644386411.432868][144031:144036] CHIP:DMG: 					Endpoint = 0x0,
[1644386411.432875][144031:144036] CHIP:DMG: 					Cluster = 0x36,
[1644386411.432879][144031:144036] CHIP:DMG: 					Attribute = 0x0000_0005,
[1644386411.432883][144031:144036] CHIP:DMG: 				}
[1644386411.432887][144031:144036] CHIP:DMG: 					
[1644386411.432892][144031:144036] CHIP:DMG: 					Data = 0, 
[1644386411.432895][144031:144036] CHIP:DMG: 			},
[1644386411.432900][144031:144036] CHIP:DMG: 			
[1644386411.432904][144031:144036] CHIP:DMG: 		},
[1644386411.432908][144031:144036] CHIP:DMG: 		
[1644386411.432912][144031:144036] CHIP:DMG: 	],
[1644386411.432917][144031:144036] CHIP:DMG: 	
[1644386411.432920][144031:144036] CHIP:DMG: 	SuppressResponse = true, 
[1644386411.432924][144031:144036] CHIP:DMG: 	InteractionModelRevision = 1
[1644386411.432927][144031:144036] CHIP:DMG: }
[1644386411.432959][144031:144036] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0036 Attribute 0x0000_0005
[1644386411.432980][144031:144036] CHIP:TOO:   BeaconLostCount: 0

```

Survensa, 2022-02-09:
@yufengwangca
             
           yeah , we are also using the linux platform and I checked that now its working fine on the latest commit ID.

somu1710, 2022-02-09:
@yufengwangca

       Yeah , we are also using the linux platform and I checked with the latest commit ID now its working fine as well.

closed 2022-02-10 by somu1710