project-chip/connectedhomeip #13430: `chip-tool` needs support for Custom Flow commissioning

liamgonyea, 2022-01-10:
#### Problem
`chip-tool` needs support for Custom Flow commissioning for test case **[TC-DD-3.9] Commissioning Flow - Custom Flow**.

Steps 1-3 all require that `chip-tool` read an Onboarding Payload from a DUT accessory and fetch data from a Distributed Compliance Ledger (DCL) to assist the user in commissioning the DUT to `chip-tool ` 

[image]

#### Test Case
**[TC-DD-3.9] Commissioning Flow - Custom Flow**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicediscovery.adoc#339-tc-dd-39-commissioning-flow---custom-flow

vivien-apple, 2022-02-08:
@raju-apple We spoke about the onboarding payload at some point. This can be done without `chip-tool` at the moment, is that correct ? If so, we need to update the priority of this issue.

woody-apple, 2022-03-03:
Given above, moving to post V1.

closed 2022-09-22 by stale[bot]