project-chip/connectedhomeip #13441: [TC-SWDIAG-1.1,TC-SWDIAG-3.1] Reading the List is not implemented in YAML framework

kowsisoundhar12, 2022-01-11:
Summary:
As per the Test-plan we have to read the list of ThreadMetrics struct  from DUT to TH but reading the list in not implemented in YAML framework
Test-Plan Screenshot:
[image]

Spec-Screenshot:
[image]

woody-apple, 2022-01-26:
YAML supports lists at this point, is this working for you now? Can you please re-open with a clear explanation of what's failing, or missing. Thanks

kowsisoundhar12, 2022-02-02:
@woody-apple
 i have ran this testscripts  again with implemented list fields but i am getting following failure response
[image]
Teststep screenshot:
[image]

Attached error logs:
[Err-log.zip](https://github.com/project-chip/connectedhomeip/files/7985295/Err-log.zip)

bzbarsky-apple, 2022-02-04:
The test is trying to read from endpoint 1.

This cluster is required to be on endpoint 0 by spec.

So it does not work.