project-chip/connectedhomeip #13447: [Commissioning] Thread + BLE: Non-concurrent mode implementation

doru91, 2022-01-11:
#### Problem
* current Matter implementation for Thread + BLE devices is using concurrent commissioning flow (see Chapter 5.5 - Commissioning Flows). This implies concurrent usage of Thread + BLE for a short period of time;
* Implementing concurrent mode is not trivial on the platform side. Also, the BLE flash requirements are around 100K and the concurrent-mode doesn't allow switching between a BLE only image and Thread only image. This could be a problem for a constraint device. 

 #### Proposed Solution
* Is there any plan for adding the non-concurrent mode into the implementation?
*
I guess that the main challenge is to send the request for the ConnectNetwork command on BLE then determine the success of this command using the operational network (Operational Discovery of the node using Thread). 

@andreilitvin @tcarmelveilleux

tcarmelveilleux, 2022-01-11:
Someone has to implement it. There is no volunteer right now as I understand it since initially it was understood to only apply to SoftAP which seemingly nobody is targetting for 1.0.

Since there is desire by some to do non-concurrent commissioning with Thread/BLE, someone who has that requirement could implement it.

The spec is very clear about the behavior, but I don't have a good view myself of the amount of work needed to orchestrate the different transitions at the Commissioner. Since this mode relies on alternate error-handling behavior , such as not expecting command responses which may not come due to change of mode of operation of the transport layer, and that it requires commissioners to be able to choose/understand which mode is at play, I am not certain it can easily be done in a common way.

andy31415, 2022-01-11:
TE8 deadline is a bit under 3 weeks away at this point, so I find it unlikely that complex changes will make it in 1.0 at this time. We need both a volunteer (likely someone familiar with working on a device directly  affected by it and with image capability swap) and development likely needs to be done outside 1.0 like a feature branch or separate repo and would be considered for integration after 1.0 release.

closed 2022-10-18 by stale[bot]

s07641069, 2022-12-27:
https://github.com/project-chip/connectedhomeip/pull/24201