project-chip/connectedhomeip #13464: Simulated Device Crashing immediately after pairing.

raju-apple, 2022-01-11:
In simulated devices immediately after pairing, the simulated device crashes. 

This is still happening even after this [PR](https://github.com/project-chip/connectedhomeip/pull/13351) merged 

Attaching logs.

Failures when trying to commission the chip-app1 app , attaching logs [Running both the apps on the same Ubuntu machine]
[chip-app1.zip](https://github.com/project-chip/connectedhomeip/files/7849366/chip-app1.zip)
[chip-tool.zip](https://github.com/project-chip/connectedhomeip/files/7849368/chip-tool.zip)

vivien-apple, 2022-01-13:
Can you provide steps to reproduce ? It seems like it just works for me.

kvikrambhat, 2022-01-13:
@vivien-apple Ensuing are the steps followed after pulling latest code
1. Generate the ZAP files, and build the chip-app1 binary completing - " ./scripts/examples/gn_build_test_example.sh app1"
2. Build all the apps [chip-tool] - "./gn_build.sh"
3. Run the chip-app1 - "./out/app1/chip-app1 --secured-device-port 5540 --discriminator 3840 --command Test_TC_DM_1_3_Simulated"

raju-apple, 2022-01-13:
@kvikrambhat Debugged this with @vivien-apple Looks like couple of things should fix it ,
Try changing these steps 

> @vivien-apple Ensuing are the steps followed after pulling latest code
> 
> 1. Generate the ZAP files, and build the chip-app1 binary - " ./scripts/examples/gn_build_test_example.sh app1"
> 2. Build everything including chip-tool - "./gn_build.sh" (  Try running this instead ./scripts/examples/gn_build_example.sh examples/chip-tool out/debug/standalone chip_config_network_layer_ble=false)
> 3. Run the chip-app1 - "./out/app1/chip-app1 --secured-device-port 5540 --discriminator 3840 --command Test_TC_DM_1_3_Simulated"
> 4. In a new terminal run the following two commands to commission the app1 -
>    a ./out/debug/standalone/chip-tool pairing ethernet 0x654321 20202021 3840 127.0.0.1 5540
>    b ./out/debug/standalone/chip-tool generalcommissioning commissioning-complete 0x654321 0 ( This step is no longed needed , the commissioning complete is done from the platform side now )

kvikrambhat, 2022-01-17:
I still see the same issue when i try the suggested changes. attached logs
[chip-.zip](https://github.com/project-chip/connectedhomeip/files/7879125/chip-.zip)

closed 2022-12-21 by stale[bot]