project-chip/connectedhomeip #13507: GroupSession needs to manually deleted after every group communication

mkardous-silabs, 2022-01-12:
#### Problem
GroupSession needs to be deleted after each group communication by calling the `RemoveGroupSession`.
https://github.com/project-chip/connectedhomeip/pull/13505#discussion_r783384089

 #### Proposed Solution
Instead of having the same logic in every location where a `GroupSession` is created, call `RemoveGroupSession` for a GroupSession inside the `SessionHandle` destructor.

bzbarsky-apple, 2022-01-12:
@kghost, can you take a look at this, please?

Note: this is a regression from #13330

jepenven-silabs, 2022-02-22:
@mkardous-silabs implemented a temporary solution by deleting everything manually. As @bzbarsky-apple said the right way would be that the GroupSession gets deleted automatically by the destructor of the sessionHandle

turon, 2022-02-25:
Couldn't the life of a `GroupSession` be tied to each underlying, active operational group key?

PR #15708 (2022-03-03): Make group session ephemeral
#### Problem
Fixes #13507

#### Change overview
Make group session ephemeral

Change the group session class into 2 classes:
* IncomingGroupSession
* OutgoingGroupSession
Although they shares save member fields, but their usage doesn't overlap, and `GetIncomingSubjectDescriptor` have different meaning for these 2 objects. Using 2 different classes can prevent people using the wrong object.

#### Testing
Passed unit-tests.