project-chip/connectedhomeip #13509: On-network commissioning does not factor in the Commissioning Mode (CM) Subtype

sagar-apple, 2022-01-12:
#### Problem

Commissioning via the CHIPDeviceController's Setup Payload based API `DeviceCommissioner::PairDevice(NodeId remoteDeviceId, const char * setUpCode)`, might fail if a device is removed from and then re-added to a Fabric. 

This is because the Device Controller is not checking the Commissioning Mode subtype before attempting to connect to a device.  When a device is forcibly removed (or if commissioning fails or is interrupted), its `matterc._udp` service might linger on DNS for a some time but with the CM=0 subtype. The commissioner currently doesn't look at the subtype and on the subsequent commissioning attempts will try and communicate with the device over DNS rather than also look for it over BLE. 

For devices that must be commissioned over BLE, this means commissioning appears unreliable. 

 #### Proposed Solution
The commissioner must ignore a CM=0 value discovered on mDNS. It should instead continue to fallback to BLE.
```
The key `CM` (Commissioning Mode) SHALL indicate whether or not the publisher of the record
is currently in Commissioning Mode and available for immediate commissioning. 
When in commissioning mode, the value associated with the `CM` key indicates the source of the passcode.

Four situations are legal:
. The absence of key `CM` SHALL imply a value of 0 (`CM=0`).
. The key/value pair `CM=0` SHALL indicate that the publisher is not currently in Commissioning Mode. 
. The key/value pair `CM=1` SHALL indicate that the publisher is currently in Commissioning Mode and requires use of a passcode for commissioning provided by the Commissionee (e.g., printed on a label or displayed on screen), such as when the device is in a factory-new state or when the <<ref_OBCW, Open Basic Commissioning Window>> command has been used to enter commissioning mode.
. The key/value pair `CM=2` SHALL indicate that the publisher is currently in Commissioning Mode and requires use of a dynamically generated passcode for commissioning corresponding to the verifier that was passed to the device using the <<ref_OCW, Open Commissioning Window>> command.
```

tcarmelveilleux, 2022-01-12:
I would argue that we could also make CM=0 less prominent. That is, unless there is a pairing hint available, and we have a product that desires manual user-entry into commissioning, and that this is enabled by some CHIP_CONFIG flags, we should not have _matterc._udp records advertised at all outside of commissioning and they should be removed after commissioning.

turon, 2022-02-01:
Recent issue related to commissioning should be fixed in v1.
Removing tag `v1_triage_split_4`

bzbarsky-apple, 2022-02-12:
@sagar-apple Did #13794 address this?

cecille, 2022-02-25:
This appears to be fixed by the linked PR (going by the thumbs up), so i am closing this issue. Please re-open if I am incorrect.

closed 2022-02-25 by cecille