project-chip/connectedhomeip #13543: Cannot specify interaction timeouts using the CHIPClusters or InvokeInteraction APIs

mrjerryjohns, 2022-01-13:
#### Problem

While the lower-level IM primitive for sending commands on the client side (`CommandSender`) permits specifying a timeout for the interaction as a whole, the higher level CHIPClusters or InvokeInteraction APIs don't.

 #### Proposed Solution

Add an extra argument to `ClusterBase::InvokeCommand` and `Controller::InvokeCommandRequest` APIs.

bzbarsky-apple, 2022-01-13:
At some point we should think a bit about whether the arguments for InvokeCommand/InvokeCommandRequest should be a struct instead of a long list....

mrjerryjohns, 2022-06-23:
This has been solved by adding `SetCommandTimeout` to `ClusterBase`, and using that to set the timeout.

closed 2022-06-23 by mrjerryjohns