project-chip/connectedhomeip #13552: Timeout value for enabling Wi-Fi network is too short in AutoCommissioner

g-coppock, 2022-01-13:
#### Problem

When commissioning an m5stack onto Wi-Fi, it may take multiple association attempts to establish a connection. In between attempts, there is a 5 second delay. Currently the [`CommissioneeDeviceProxy`](https://github.com/project-chip/connectedhomeip/blob/master/src/controller/CommissioneeDeviceProxy.cpp#L85) delegates to [`CHIPClusters`](https://github.com/project-chip/connectedhomeip/blob/master/zzz_generated/controller-clusters/zap-generated/CHIPClusters.cpp#L4764) to call `ConnectNetwork`, and this ends up using the default 12 second timeout in [`InteractionModelDelegate`](https://github.com/project-chip/connectedhomeip/blob/master/src/app/InteractionModelDelegate.h#L39). If there is a retry in Wi-Fi, then in most cases the command will time out early before we receive the successful connection message, causing commissioning to fail.

```
2022-01-12 15:37:11.210 19995-20139/com.google.android.gms D/DMG: Time out! failed to receive invoke command response from Exchange: 54520i
2022-01-12 15:37:11.210 19995-20139/com.google.android.gms D/ZCL: DefaultResponse:
2022-01-12 15:37:11.210 19995-20139/com.google.android.gms D/ZCL:   Transaction: 0x7961fff3d0
2022-01-12 15:37:11.210 19995-20139/com.google.android.gms D/ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
2022-01-12 15:37:11.210 19995-20139/com.google.android.gms D/CTL: Received failure response 1
2022-01-12 15:37:11.217 19995-20139/com.google.android.gms V/ChipController: onStatusUpdate(1) [CONTEXT forced=true service_id=288 ]
```

 #### Proposed Solution
Long term, the suggested timeout attributes accessible via the Network Commissioning Cluster should be used, but short term perhaps a more generous timeout (~30s?) should be used.

g-coppock, 2022-01-13:
See also https://github.com/project-chip/connectedhomeip/issues/13543

cecille, 2022-02-25:
The timeout is now 30s, added https://github.com/project-chip/connectedhomeip/issues/15611 to track reading the correct timeout from the device.

closed 2022-02-25 by cecille