project-chip/connectedhomeip #13594: Add support for null'ing out fabric-sensitive data in cluster-objects

mrjerryjohns, 2022-01-14:
#### Problem

When encoding a fabric-scoped cluster object with fabric-sensitive fields in it, those should be nulled out if the accessing fabric doesn't match the fabric-index value within that object.

mrjerryjohns, 2022-01-14:
To reduce the scope of this change, suggestion is to add an extra `Optional` argument to the existing templated `Encode(TLV::TLVWriter & writer, TLV::Tag tag, const X & x)` (and its optional and nullable variants) that accepts an optional fabric index.

The API would default initialize that to not contain a fabric index so that existing call-sites in clusters don't have to change.

In the generated cluster object code, for fabric-scoped clusters with fabric sensitive fields, it should validate that there is indeed a non-optional value provided (and error out appropriately) and correctly omit sensitive fields.

mrjerryjohns, 2022-01-14:
Slight pivot:

We should do some kind of SFINAE magic here to create two different versions of the Encode APIs such that any objects that have fabric-sensitive fields will require the API containing a non-optional fabric index argument. This ensures that we can catch errors at compile time rather than at run-time.

bzbarsky-apple, 2022-01-14:
So basic shape of the API would then look as follows:
1. A struct that has fabric-sensitive fields would have an Encode that takes an additional FabricIndex arg (not an Optional or anything).
2. A struct that does not have such fields would have that, but also have an API that does not take a FabricIndex and just calls the one that does with FabricIndex set to 0.
3. We would have a version of DataModel::Encode that takes fabric index and requires the extra arg, and a version that does not take a fabric index and does not require the extra arg.
4. For optional/nullable we have two DataModel::Encode signatures, with and without fabric index.
5. For everything else we can just have DataModel::Encode take an optional FabricIndex, defaulting to 0 (which is then ignored).

Sound about right?

mrjerryjohns, 2022-01-18:
As per discussion, we can omit 2).

closed 2022-03-04 by mrjerryjohns