project-chip/connectedhomeip #13595: Generated cluster-objects code has incorrect field ids

bzbarsky-apple, 2022-01-14:
#### Problem
The spec sometimes starts field id numbering at 1 and sometimes at 0.

Our generated cluster-objects code starts the numbering at 0 for command args, but at 1 for struct/event members.  In various cases our generated code uses field ids that do not match the spec.

For struct/event members, we allow the XML to override the numbering, but not in consistent ways.  For events its `id="..."` on the field, for structs it's `fieldId="...."` on the field.

For command args we also support `fieldId="..."` but don't expose the result until https://github.com/project-chip/zap/pull/372 is fixed, so cluster-objects is not using it.

 #### Proposed Solution
Once https://github.com/project-chip/zap/pull/372 lands:

* Decide whether it's `id=` or `fieldId=`.
* Audit all field ids to make sure they match the spec.  Ideally by explicitly specifying them all, instead of relying on the default behavior (which is 1-based numbering).  The output of https://github.com/project-chip/connectedhomeip/pull/13544 might help here.

mrjerryjohns, 2022-02-28:
> but at 1 for struct/event members

This has now been fixed to start to 0.

bzbarsky-apple, 2022-09-06:
I think at this point all the numbering is consistent.

closed 2022-09-06 by bzbarsky-apple